flipperdevices/flipperzero-firmware #1715: Eavesdrop ISO 14443 Type A communication between reader and card and show relevant values

koalazak, 2022-09-07:
### Description of the feature you're suggesting.

It would be great a new option to sniff the communication between reader and external tags to then show anti-col and authentication encrypted but relevant values: `uid, tag challenge, reader challenge, reader response, tag response`

Similar to `hf 14a snoop` + `hf list 14a` proxmark's commands.

I imagine this in two steps:

1. Actual sniff and save the data. 
    NFC > Extra Actions > Eavesdrop
    Message "Place Flipper and tag over the Reader..." and "REC" button on screen to start recording.
    Stop button appears and once clicked "save>" button appears. 
    Same behavior as "Read Raw" in sub-ghz menu. 

2. Select the dump and click on "Auth Info" to show the hex values for `uid, tag challenge, reader challenge, reader response and tag response` for all authentication flows in the dump.
     NFC > Saved > Dump1 > Auth Info
     Show the 5 relevant values on screen

[image]

With this values you can crack the keys externally without brute force them. What do you think?

### Anything else?

Is this currently available on FZ but inside another feature I missed? (I mean the sniff part)
The current FZ tag brutefoce attack is based on dictionary or full scan?
Please let me know.

Thank you for your awesome work

yeldiRium, 2022-10-06:
@koalazak The current FZ tag analysis is based on this dictionary: https://github.com/flipperdevices/flipperzero-firmware/blob/dev/assets/resources/nfc/assets/mf_classic_dict.nfc

koalazak, 2022-11-27:
hello @skotopes,
how hard is implement this? or at least sniff and dump the card<>reader communication (and extrach those values manually later)
Do you know if somebody is working on this? or if this is the backlog?

thank you!

skotopes, 2022-11-27:
@koalazak partially possible, quite complex.

What you want is to simultaneously capture upstream and downstream which normally got different type of modulation/coding. This feature requires transceiver in bypass modem mode or SDR(like in proxmark). 
In theory st25r3916 do have transparent mode and special debug pins that can do something similar. But debug pins are not wired to MCU so you can not use them. As for transparent mode it requires some experiments and testing. Also it will be computationally intensive.
In theory there is a third way: some kind of hybrid mode where you using transceiver modem for capture.

skotopes, 2022-11-27:
And yes It's in backlog.

sfjuocekr, 2023-12-23:
What if one just wanted to eavesdrop on the communication from the reader/writer and not the card?

For example you have a NFC card of which you have all the keys but one, this card is not vulnerable to any current attack vectors and so you would need to sniff the key as used by the reader to access the card.

Essentially, just sniffing the keys as sent to the card.

skotopes, 2024-01-03:
@gornekich is working on sniffing

koalazak, 2024-06-18:
hello @gornekich thanks for taking care of this. Is there any branch I can test?

thank you

Laimutis, 2024-06-25:
+1 interested in this feature

AkechiShiro, 2024-06-25:
Is there any news since January @skotopes ? We haven't seen much, is there a draft PR where @gornekich is working ?

skotopes, 2024-06-25:
It's in progress, but there is no timeline.

MaggieK410, 2024-12-20:
Is there any progress since June on this @skotopes ? Im not sure if I maybe couldnt find the implementation yet, or if its still in progress