flipperdevices/flipperzero-firmware #1718: Sort Mifare Dictionaries

tkerby, 2022-09-07:
### Description of the feature you're suggesting.

The Mifare dictionaries are large and slow to read cards. I'd like to see a mechanism to sort the dictionaries, so the most common keys appear at the top of the list.

A few thoughts on how we might accomplish this

1. Implement a front end to a web service running mfkey32 within the phone apps. Add the keys to the dictionary or log the frequency whenever someone submits nonces. Sort based on frequency.
2. Allow the app to anonymise and submit cache files to a server (i.e. remove the UID and data - just send keys). Identify the frequency of keys appearing and add any new ones to the dictionary once they reach a threshold. Sort by frequency
3. On flipper, move cached keys to the top of the dictionary or implement a function to add them to a user dictionary. I often have one card I can take the time to read, but I then want to copy others. At the moment, that's hard to do for speed. 

### Anything else?

Another useful feature here would be a 'read similar' function. I.e. allow someone to select a saved card and use its cached keys to read other cards before trying the dictionary.

zviratko, 2022-09-09:
This is somewhat similiar to my open issue
https://github.com/flipperdevices/flipperzero-firmware/issues/1646

It sounds like the answers in the issue are one way of looking at it and arguably correct (or more correct than my proposed solution). Maybe only the docs are lacking so we don't know how to properly use it? Of course having a UI setting for that would be very nice, otherwise it's always going to be "one size fits all" - some people are going to regularly attempt to crack encrypted cards, some people mostly deal with default keys (and thus would prefer having those on top), other people will read their own cards with their own keys and construct their own dictionaries...

I haven't yet dug too deeply into it, also it might differ between firmware flavours. My unfounded suspicion (since I haven't really spent more time on it) is that some firmwares include "user" dicts which renders them unsuitable for the use ofw authors had in mind.

skotopes, 2023-03-13:
User dict should solve this issue.

closed 2023-03-13 by skotopes