flipperdevices/flipperzero-firmware #1720: Generic Keeloq 64 Bit with 2 buttons pair

BrunoTCouto, 2022-09-07:
### Description of the feature you're suggesting.

Hi !  I have a **generic keeloq 64bit remote**, that’s is the most common over here in Brazil. 

I tried paring my flipper with the receiver using the DoorHan_433 but the receiver asks for pressing two buttons at the same time, so I was unable to pair.

Speaking with @skotopes on discord he asked me to open an issue here in github for implementation.

## Useful pictures:
### Sub-GHz Captures by Flipper
<details>

[image]
[image]
[image]

</details>

### Remote
<details>

[image]
[image]
[image]
[image]

</details>

### Receiver
<details>

>   [image]
>   **🇧🇷 Pressione duas vezes B1+B2 de um TX**
>   **🇺🇸 Press two times B1+B2 of one TX**

  

>   [image]
>   **🇧🇷 Pressione duas vezes B1+B2 de um TX**
>   **🇺🇸 Press two times B1+B2 of one TX**

</details>

Skorpionm, 2022-09-08:
your system is not Dorhan, but something like BFT. Flipper does not save or transmit dynamic protocols, Dorhan exception

BrunoTCouto, 2022-09-08:
So flipper will never be able to pair with gates that are dynamic protocols?

I wasn't thinking of cloning my controller, but using the flipper instead of my current controller

Should I close this issue ?

skotopes, 2023-06-22:
Yep. I think we can close it. We don't know MF key for this system.

closed 2023-06-22 by skotopes