flipperdevices/flipperzero-firmware #1721: NFC Emulation not working anymore

Jay4730, 2022-09-08:
### Describe the bug.

It was working a few months ago. I'm not sure if it's a hardware issue due to my flipper falling hard enough to crack the screen; I didn't need to emulate for a few months after. The only thing I have to test it out on is with amiibo on my switch but I've tried with other systems/official controllers as well and nothings worked. I've tried different firmwares, amiibo file dumps, restoring my flipper and nothings changed. It can still read tags from my desk mat and my credit card just fine.

### Reproduction

Read nfc tag
Emulate tag
Nothing happens

gornekich, 2022-09-08:
Hello @Jay4730 
Could you please emulate attached file and read it with you mobile.
[test.nfc.zip](https://github.com/flipperdevices/flipperzero-firmware/files/9526486/test.nfc.zip)

Jay4730, 2022-09-08:
I saved it but my phone doesn't read it, nothing happens

Crsarmv7l, 2022-09-13:
My flipper also doesnt emulate nfc. I have never dropped it so I doubt that is the issue

Jay4730, 2022-10-03:
No worries on the late reply, I just updated to the newest fw and still nothing

apollo13, 2022-11-07:
Hi, I am having the same problem. The `test.nfc` file doesn't emulate properly either. This used to work and I can still scan cards, but no matter what I emulate no device detects it. Can I enable debug logging or so?

apollo13, 2022-11-07:
Trace log shows:
```
12489191 [I][LoaderSrv] Starting: NFC
12489196 [I][AnimationManager] Unload animation 'L1_Recording_128x51'
12489217 [T][StorageAPI] File/Dir 0000C358 alloc
12489230 [T][StorageAPI] File 0000C358 - 0000D2EC open (/any/nfc/test.nfc)
12490070 [T][StorageAPI] File 0000C358 - 0000D2EC closed
12490073 [T][StorageAPI] File/Dir 0000C358 free
12490081 [D][DolphinState] icounter 168, butthurt 0
12490089 [D][MfUltralight] Prepare emulation
```
and then nothing

apollo13, 2022-11-07:
Correction, the test.nfc now works again after a reboot. I think I am running into #1768 instead.

skotopes, 2022-12-01:
I'll close this issue then

closed 2022-12-01 by skotopes