flipperdevices/flipperzero-firmware #1726: PRNG error making Emulated NFC tags unreadeable

shelld0n, 2022-09-09:
### Describe the bug.

while trying to debug a failing NFC emulation i came accross a small behavior i would like to share, maybe it can help making the NFC emulation a little bit better.

I have an NXP MIFARE Classic 1k tag, i read it with the flipper and it succeed with all keys found and all sectors read. now when i emulate it on the reader it does not work. 

so i decided to use a proxmark3 to see  the difference between what the flipper emulates and whats the real tag is sending. and here is the results: 

This is when i read the REAL TAG with proxmark3 : 
```proxmark3> hf search

 UID : 5a 2f 1c 93
ATQA : 00 04
 SAK : 08 [2]
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1
proprietary non iso14443-4 card found, RATS not supported
No chinese magic backdoor command detected
Prng detection: WEAK

Valid ISO14443A Tag Found - Quiting Search

proxmark3>
```

And this is what proxmark3 sees when i emulate with the flipper: 
```proxmark3> hf search

 UID : 5a 2f 1c 93
ATQA : 00 04
 SAK : 08 [2]
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1
proprietary non iso14443-4 card found, RATS not supported
No chinese magic backdoor command detected
PRNG error: selecting tag failed, can't detect prng.
Prng detection error.

Valid ISO14443A Tag Found - Quiting Search

proxmark3>
```
we can notice that there is no difference except for the PRNG error , it says PRNG detection errors,  and for now i can't say why we have this difference , is it a normal thing when emulating  ?
i think the PRNG error is whats causing the emulation to fail on some readers.

### Reproduction

1 . Read a TAG with the Flipper and make sure you have all keys and sectors
2 . Read the same TAG with proxmark, and save the output (foucs on the PRNG part)
3 . Emulate the same TAG with the flipper on the proxmark reader and save the output (foucs on the PRNG part)
4 . You should notice that the emulated TAG with the Flipper results in a PRNG error while the Real TAG is not.

Astrrra, 2022-09-09:
Hi, thanks for your feedback! Can you please provide the trace files of both reading attemps? They can be saved by running `trace save -f <filename>` right after the search command.

shelld0n, 2022-09-10:
Sure, here is the Two trace files where : 

nfctag-1 is the trace of the NFC tag 
flipper-1 is the trace of the flipper emulation 

note sure if it can help, thank you for the effort :) 

[trace_files.zip](https://github.com/flipperdevices/flipperzero-firmware/files/9541538/trace_files.zip)

skotopes, 2022-12-01:
Please reopen this issue If you still experiencing this problem

closed 2022-12-01 by skotopes