flipperdevices/flipperzero-firmware #1727: Flipper reports firmware update after empty battery

jkctech, 2022-09-10:
### Describe the bug.

When plugged into my PC after an empty battery, Flipper turns on and displays the message "Firmware pdate success"
[image]

### Reproduction

1. Drain battery :)
2. Plug in flipper (Don't know if PC or charger makes difference)
3. Watch the update message as Flipper turns back on

GMMan, 2022-10-07:
Can you reproduce this more than once? From the codepaths it seems it can only show if a file exists at `/int/.slideshow`, and that file contains the frames to display. If the file does not exist, this cannot be displayed. So if the file is present, most likely you powered off the device/it ran out of battery after an update and before you acknowledged the message, or LFS failed to delete the file due to being out of space or something like that, which may happen if you're using an older debug firmware or custom firmwares that don't watch the amount of flash space they're using.

jkctech, 2022-10-07:
I remember at that time I was able to reproduce this at least once more, which was the picture shown in the report, but I pulled out the usb cable the before acknowledging, it immediately died because of the battery and on replug showed the message again (which makes sense reading your above message)

The flipper dieing before the ack of the update is also out of the question for me since I play around after every update and it was at least a few days after the update.

I have never run development builds and at that time and I was 100% up to date with the latest release since I look for new releases daily, oops 😂

So I am not sure what things are at play here, anything else I could have a look at?

GMMan, 2022-10-07:
See if you can somehow reproduce this again, but before acknowledging the message, open up qFlipper and see if there's a `.slideshow` file in internal. Then acknowledge, see if it disappears. If it doesn't disappear, you'll probably want to enable logging and check if there's any errors deleting the file. If it does disappear, replug again and see if you get the screen. If you do, that might be some indications that parts of the updater is running when it's not supposed to.

skotopes, 2022-10-08:
This issue was fixed in previous release.

closed 2022-10-08 by skotopes