flipperdevices/flipperzero-firmware #1730: Mifare DESfire saves being but inaccessible due to "Can not parse file"

Masqueey, 2022-09-11:
### Describe the bug.

When trying to read previously saved Mifare DESfire cards, I get a "Can not parse file" error. However, when I try to access the saved data through the mobile app, it works fine and can even get the Flipper to emulate it. The flipper is also able to emulate the card when it has just been read and not yet saved.

### Reproduction

1. Go to NFC -> Read
2. Read a Mifare DESfire
3. Save it
4. Go to NFC -> Saved
5. Select the Mifare DESfire
6. "Cannot parse file"

### Target

Version 0.66.1

### Logs

```Text
n/a
```

### Anything else?

Let me know how to get the right logs if you need them to solve this issue

djsime1, 2022-09-20:
I don't have any DESFire cards to test this on, however I believe Flipper only stores basic info about the card (like UID). It's impossible to fully save the contents of a DESFire card.

Masqueey, 2022-09-21:
I have done some research myself and came to the same conclusion; part of the information is stored on an inaccessible part of the chip. (Inaccessible from the outside that is.) This issue can be closed.

closed 2022-09-21 by Masqueey

Axlbuld, 2022-11-16:
Same with current release and MiFare DesFire EV2. Trying to store the UID, only, but save failes.