flipperdevices/flipperzero-firmware #1732: Allow signal replay directly from the SubGhz Read screen

cusspvz, 2022-09-13:
### Description of the feature you're suggesting.

Adding the ability to read Raw signals directly from the Read screen was a great idea. Although, testing if the signal was correctly decoded by Flipper you got to first save it, then loose all the other signals that you might have captured and then “send”/replay the signal.

I wonder how easy would it be to add up the send action to the signal info screen directly.

Skorpionm, 2022-09-14:
all received signals in Read , for which transmission is allowed, can be transmitted immediately without being stored, and the rest of the history is not lost.  Or am I misunderstanding what you want

0nnyx, 2022-09-17:
OP is right. If you select Read, then Config and enable Detect Raw, the Raw captures can only be saved and not directly replayed unlike non-raw captures.

Skorpionm, 2022-09-17:
only official firmware is discussed here

cusspvz, 2022-09-20:
Just figured out that the Read Raw feature is not part of the official firmware. My bad, closing the issue. Thanks @Skorpionm

closed 2022-09-20 by cusspvz