flipperdevices/flipperzero-firmware #1737: Dummy Reader mode, to power passive 13.56mhz devices

eriequiet, 2022-09-15:
### Description of the feature you're suggesting.

a lot of biohackers have 13.56mhz implants that possess a led, 
the flipper does a great job at pushing enough field to power these, but it immediately reads the tag and stops the field 
frustrating if you are trying to show off the LED underneath your skin
a mode to simply power the field, and not react to any tags would be super useful

djsime1, 2022-09-20:
I believe this is already possible, first enable debug menus by changing `Settings > System > Debug` to ON, then open the new "Debug" menu in the NFC app and select Field. The warning text is mostly there because it drains the battery quicker, leaving the field on won't damage the device.

Astrrra, 2022-12-08:
As @djsime1 said, this has already been implemented. 

However, leaving the field on for a long time (more than 20 minutes) actually can damage your device, as the ST25R3916 NFC chip is located directly under the battery, and leaving it on causes it to heat up quite substantially, which unevenly heats up the battery, which is bad for its health and longevity.

closed 2022-12-08 by Astrrra