flipperdevices/flipperzero-firmware #1741: Frequency Analyzer Log

pstray, 2022-09-16:
### Description of the feature you're suggesting.

It would be really useful if the Frequency Analyzer of the Sub-GHz app could have a log of the frequencies it saw.

Just showing the frequency while the signal is strong enough often causes it to show it for less than a second. It would be nice to have a list of the frequencies found, and maybe show that last one on the front screen.

skotopes, 2022-10-16:
Available in dev and will be release in 0.69

closed 2022-10-16 by skotopes

michaelfaith84, 2022-10-21:
How about a "log to file" option that makes a csv with timestamps etc?

Bedrovelsen, 2023-03-07:
> How about a "log to file" option that makes a csv with timestamps etc?

Yes please. I mean, I can try add this as it was the first thing I noticed when I got my flipper this week, not being an obvious or implemented option even though its a one line change to add auto output to file pre exit.