flipperdevices/flipperzero-firmware #1743: Implement default meaningful filename when saving captures

0nnyx, 2022-09-17:
### Description of the feature you're suggesting.

Problem: The current default filenames have few (maybe 5 maximum) "funny" names which don't mean anything relevant. The flipper clickable buttons are very impractical to change these filenames.

Proposed solution: Remove the default funny names and implement a meaningful standard timestamp YYYYMMDD-HHMMSS +  capture type + frequency. Possible example:  20220917-123030-RAW-434Mhz

djsime1, 2022-09-20:
I could see this being added as a setting possibly...

Skorpionm, 2022-09-23:
https://github.com/flipperdevices/flipperzero-firmware/pull/1772
check

0nnyx, 2022-09-23:
That PR is a good first step. However, I was rather thinking about doing this sytemwide for all capture types, meaning RFID, NFC, SubGhz and potentially for fap allowing file saving.

wdoekes, 2022-10-09:
I would rather have it reuse the previously saved filename and add a number to it. That way I could do multiple recordings, and maybe only change the last digits of the filename to store something meaningful.

Just a thought.

(The timestamps are available on the filesystem, so if you take the files to a computer using the sdcard, you can view them there.)

skotopes, 2022-10-16:
One of limitations that we have is a small screen. Putting everything into file name will make it unreadable on device.
We'll think about changing naming scheme system wide.

ppseprus, 2022-10-22:
I actually wanted to open a PR where under Settings, one would be able to switch between file name types for all applications. Unfortunately, neither my C / C++ skills, nor the amount of my free time allowed this to happen.

However, I would suggest to go into a more customisable direction with such things: allow users to switch between different degrees of precision. From random to timestamp with protocol and what not...

#opinionated

closed 2022-12-01 by skotopes