flipperdevices/flipperzero-firmware #1747: FAP : Unsatisfied imports

csBlueChip, 2022-09-18:
### Describe the bug.

When I run my (previously working plugin) as a FAP, I get :
```
[W][elf_hashtable] Cant find symbol 'LL_TIM_DeInit' (hash d462ee02)!
[W][elf_hashtable] Cant find symbol 'calloc' (hash f5e616f3)!
[E][fap_loader_app] FAP Loader failed to map to memory /ext/apps/Cyborg Systems/bc_pwm_demo.fap: Found unsatisfied imports
```
The `calloc()` is easy to workaround: {`malloc`, `memset`}

...But the lack of an export for `LL_TIM_DeInit()` is not so obvious:
https://github.com/csBlueChip/FlipperZero_plugin_PWM/blob/main/bc_pwm.c#L40

This GPIO Pin configuration-strategy was lifted directly from `targets/f7/furi_hal/furi_hal_speaker.c`

### Reproduction

```
cd /path/to/flipper/
mkdir applications_user/bc_pwm_demo
git clone https://github.com/csBlueChip/FlipperZero_plugin_PWM.git
./fbt launch_app APPSRC=applications/bc_pwm_demo
```
* FZ: Exit the fap
* FZ: settings->system->log level = Warning
* PC: start putty/minicom/whatever
* PC: log
* FZ: Restart the fap
* PC: watch the error log

### Target

FZ github head (as of date of post)

### Logs

```Text
[W][elf_hashtable] Cant find symbol 'LL_TIM_DeInit' (hash d462ee02)!
[W][elf_hashtable] Cant find symbol 'calloc' (hash f5e616f3)!
[E][fapp-i] Error relocating section '.text'
[E][fap_loader_app] FAP Loader failed to map to memory /ext/apps/Cyborg Systems/bc_pwm_demo.fap: Found unsatisfied imports
```
```

### Anything else?

Please provide exports for `calloc()` the LowLevel interface.

skotopes, 2022-09-21:
Do you experience this issue in latest RC?

hedger, 2022-09-21:
calloc() is redundant, since our malloc() implementation goes beyond specs and fills allocated block with zeroes.
As for LL_* APIs, we'll expose them in the upcoming RC.

hedger, 2022-09-21:
Please check out 0.67-RC1, we've added more LL APIs to symbol table.

closed 2022-09-21 by hedger

csBlueChip, 2022-09-23:
> ... our malloc() implementation goes beyond specs and fills allocated block with zeroes ...

https://linux.die.net/man/3/malloc
"The memory is not initialized" [in italics]

It's very unlikely to become relevant for Flipper code, but do be aware that changing the way memory is initialised (or not) can have critical [security] ramifications, doing in "standard" libraries [ie. stdlib] might make arising issues very hard to spot.
https://www.schneier.com/blog/archives/2008/05/random_number_b.html

csBlueChip, 2022-09-23:
> Please check out 0.67-RC1, we've added more LL APIs to symbol table.

I found the csv, and added in the exports that I needed.  I will revert the file before I do my next pull.

Thanks.