flipperdevices/flipperzero-firmware #1753: FAP : Remote exit

csBlueChip, 2022-09-20:
### Description of the feature you're suggesting.

Issue:
If you are (a) in the Applications menu or (b) running a FAP - when you execute `./fbt launch_app APPSRC=applications/$1` the upload fails with the error `[ERROR] Unexpected response: Can't start, furi application is running`

Possible solutions:
* The upload function instructs the FZ (¿fap_loader?) to terminate the Application [Menu]
* The upload function sends a message to the FAP requesting that it exits

hedger, 2022-09-21:
Unfortunately, there's no reliable way to terminate an arbitrary RTOS thread without leaking that thread's allocated resources.
And currently there's no convention on sending termination request to an application - all of them implement that in different ways.

But, we have ways to send button inputs over RPC, so you can try emulating several "back" button presses and hope your application handles it correctly.

closed 2022-09-21 by hedger

DCNick3, 2022-10-28:
You can also issue a reboot which will obviously terminate your app and give (kind of) clean slate