flipperdevices/flipperzero-firmware #1760: Channel Island Regions should be UK frequencies for Sub-GHz

Ads-Decadence, 2022-09-20:
### Describe the bug.

Good day peeps, I have an issue I live in Jersey Channel Islands United Kingdom, flipper hardware provisioned marks me as JE (this is correct) but only gives me radio frequencies available to R03, not the UK.
I believe any users in Gurnsey (or the other Channel Islands) will have the same issue

Yours Ads 

### Reproduction

updating via a Channel island VPN?

### Target

Sub-GHz

aprosvetova, 2022-09-21:
Hi,

Please ignore R01/R02/R03 in the device info. This is legacy and doesn’t mean anything. If it says JE, you have frequencies unlocked that are allowed for JE. 

Or are you saying these actual frequencies don’t match with the local law?

aprosvetova, 2022-09-21:
Oh, I just checked that we don’t have specific config for JE, so it falls back to the default region.

Do you have any legal proof other frequencies should be unlocked too, so we could reconsider the config?

skotopes, 2022-09-21:
@aprosvetova they are some kind of Europe, I think it will be safe to assume that they got same regulations as EU. Fallback to JP is kinda strange.

Ads-Decadence, 2022-09-21:
https://www.ofcom.org.uk/ Ofcom is the Regulatory body of the United Kingdom and the Channel Islands 

Jersey Legal Information Board provides copies of local law and cites Ofcom https://www.jerseylaw.je/

skotopes, 2022-09-21:
I've added JE region data same as EU/GB. Please update your flipper.

closed 2022-09-21 by skotopes

skotopes, 2022-09-21:
We'll revise this table later with certification laboratory and add all existing countries.