flipperdevices/flipperzero-firmware #1763: Infrared: RAW format limit appears to be very low

Tylermarques, 2022-09-21:
### Describe the bug.

When attempting to send an IR signal using the RAW format, there appears to be a limit on the length of the timing allowed, and it appears to be quite low. The below IR raw signal does not show on the flipper when uploaded to the Flipper.

```
Filetype: IR signals file
Version: 1
# 
name: Garage
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 750 50 3150 50 3950 50 750 50 3150 50 3950 50 3950 50 750 50 3150 50 3950 50 750 50 3150 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 750 50 3150 50 3950 50 750 50 3150 50 3950 50 750 50 3150 50 3950 50 750 50 3150 50 750 50 3150 50 3950 50 3950 50 750 50 3150 50 750 50 3150 50 3950 50 3950 50 3950 50 3950 50 3950 50 750 50 3150 50 3950 50 3950 50 750 50 3150 50 3950 50 3950 50 750 50 3150 
```

I did some looking into the firmware and I believe the amount of timings allowed is dictated by this line in [infrared_signal.c](https://github.com/flipperdevices/flipperzero-firmware/blob/ce7b943793926ba7cffa914942f91c8791396595/applications/infrared/infrared_signal.c#L74). It appears `MAX_TIMINGS_AMOUNT` is limiting whether the signal is valid or not based on it's length. I was able to adjust this to `8096` and the flipper would then accept a larger chunk of the signal. 

### Reproduction

1. Upload the above .ir file to the Flipper
2. Navigate the flipper menu to that file
3. Observe that the button for this (named garage in this case) does not appear.

Tylermarques, 2022-09-21:
@gsurkov I'm unclear on the limits of `MAX_TIMINGS_AMOUNT` and whether or not increasing that value is the correct way to go about this. If you have a better solution, I'm all ears.

gsurkov, 2022-09-21:
Could you please specify which firmware version are you using?
The file you have provided has only 140 timings, which is nowhere near the limit.
I have tested it with current release and development branches, and it can be loaded and transmitted without problems.

The rationale behind this limit stems from the fact that whenever a remote is loaded, its signals get also loaded into the memory all at once. This could be a lot, so the maximum timing amount was set to a reasonable limit just in case.

Tylermarques, 2022-09-21:
I'm using dev, compiled on my machine using the `./fbt` command. I've attached the entire file to this message, possible my headers are incorrect? 

Thank you for the rational, I did not realize that. 

```
Filetype: IR signals file
Version: 1
# 
name: Ty_condo
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 750 50 3150 50 3950 50 750 50 3150 50 3950 50 3950 50 750 50 3150 50 3950 50 750 50 3150 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 3950 50 750 50 3150 50 3950 50 750 50 3150 50 3950 50 750 50 3150 50 3950 50 750 50 3150 50 750 50 3150 50 3950 50 3950 50 750 50 3150 50 750 50 3150 50 3950 50 3950 50 3950 50 3950 50 3950 50 750 50 3150 50 3950 50 3950 50 750 50 3150 50 3950 50 3950 50 750 50 3150 
#
name: Kaylee
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 174 22503 177 109445 176
```

Tylermarques, 2022-09-21:
Okay below are the steps I have taken, in order.
1. `sudo rm -r flipperzero-firmware`
2. `git clone --rescursive git@github.com:flipperdevices/flipperzero-firmware.git`
3. `cd flipperzero-firmware`
4. `git checkout dev`
5. `./fbt`
6. Connect the Flipper
7. `./fbt firmware_flash`

The above firmware flash fails for me. I'm presented with the error `Error: open failed` see below for full error,

```
❯ ./fbt firmware_flash
        SDKCHK  firmware/targets/f7/api_symbols.csv
API version 1.7 is up to date
openocd -f interface/stlink.cfg -c "transport select hla_swd" -f debug/stm32wbx.cfg -c "stm32wbx.cpu configure -rtos auto" -c init -c "program build/f7-firmware-D/firmware.bin reset exit 0x8000000"
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2021-12-07-17:30)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
Info : clock speed 24000 kHz
Error: open failed

scons: *** [build/oocd-firmware-flash.flag] Error 1
********** ERRORS **********
Failed building build/oocd-firmware-flash.flag: Error 1
```

I take this to mean there is likely a permissions error on the flash, but my user has access to the flipper (runing `cat /dev/ttyACM0` shows the flipper content`). I'm unsure what the issue is here, but flashing from qFlipper using the .dfu file completes successfully.

8. Open qFlipper
9. Flash Flipper using `.dfu` file found in `./build/f7-firmware-D`
10. Use flipper to navigate to `test_full.ir` 
11. No button is shown

Tylermarques, 2022-09-21:
Update: I was able to upload to my Flipper using the `./fbt flash_usb` option

gsurkov, 2022-09-21:
`firmware_flash` works only when you have the st-link connected. My bad for assuming you had it (since you mentioned building the firmware yourself).
In that case, the easier way of getting development firmware is by selecting "Development" as the firmware update channel in qFlipper. You may use that to make absolutely sure that the correct firmware is installed. `flash_usb`  you have mentioned is also an option.

Please answer to my last 2 questions as well.
Also make sure that the `test_full.ir` file that you have on Flipper's SD card contains no new line after "data:"

Tylermarques, 2022-09-21:
Sorry didn't mean to omit the last two questions. 

- I can read the file back successfully, using qFlipper to save it to a separate location under a different name. The file remains the same as above. If I remove a large section of the `data` part of `Ty_condo` button, both buttons display. 
- I'm using `git clone git@github.com:flipperdevices/flipperzero-firmware.git` so I'm fairly certain it is the official repo.

Tylermarques, 2022-09-21:
I have found the issue. There was a space at the end of multiple lines, which the flipper does not apparently care for. Removing all of them made the buttons appear

Tylermarques, 2022-09-21:
Thank you @gsurkov for helping me troubleshoot. I didn't realize that the whitespace would cause such an issue. I'm not sure whether this is a bug or not, but it is good to know that the whitespace character cannot be present there.

gsurkov, 2022-09-21:
I was able to reproduce it by adding the trailing space on line 8. It was probably automatically trimmed by the editor, since I copy-pasted it. Live and learn.

The situation looks rather buggy to me, since trailing space shouldn't really matter here. It is noted and the fix will soon be available.

For now, closing this issue since it's not really related to the maximum timing count limit.

closed 2022-09-21 by gsurkov

Tylermarques, 2022-09-21:
That works for me. Sorry for the round-about debugging but I do appreciate the help you were able to provide. Thanks for all your efforts on this project :)

gsurkov, 2022-09-21:
You're welcome ;)