flipperdevices/flipperzero-firmware #1768: NFC 14a emulation bug

gloglas, 2022-09-22:
### Describe the bug.

NFC works as should but after scanning 14a card and emulating UID, emulating no longer works.
If you exit back to NFC menu or home screen and try to emulate one of your saved cards, it doesn't work.
If you try to read a card, it will read as it should.
The NFC emulation just stops working until flipper restarts.

### Reproduction

Try NFC funcionality, emulate something.
Everything works.

Scan this type of card:
```
Tag type: ISO 14443-4
NXP - Mifare DESFire EV1 8k

Technologies Available:
IsoDep, NfcA, NdefFormatable

Serial number 04:5B:2A:C1:19:5B:04 # random serial num
ATQA 0x0344
SAK 0x20
ATS 0x80
```

Go to More > Emulate UID *press OK*
Emulation starts, yet nothing cannot be read with a reader (meaning another device).
Exit NFC app
Go back in NFC app and try to emulate anything.
It will not emulate, until you reboot os.

### Target

0.66.1 (even RC version)

### Anything else?

I think it might be related to flipperdevices/flipperzero-firmware#1726

gornekich, 2022-09-29:
Reproduced this bug, working on it. Thank you!

apollo13, 2022-11-07:
I think I am running into the same bug with a 4k MIFARE DESFire EV1, that said for me it seems that I can no longer emulate anything after simply reading the tag. That is reading this tag and then emulating the test.nfc from #1721 also doesn't work without a reboot.

apollo13, 2022-11-07:
Reading the card yielded:
```
477611 [I][NfcWorker] Mifare DESFire detected
477617 [T][StorageAPI] File/Dir 000118F0 alloc
477625 [T][StorageAPI] File 000118F0 - 0001197C open (/ext/nfc/debug.txt)
477632 [T][FuriHalNfc] State change 0 -> 10
477765 [T][FuriHalNfc] State change 10 -> 11
477798 [T][FuriHalNfc] State change 11 -> 13
477807 [T][FuriHalNfc] State change 13 -> 30
477830 [W][MifareDESFire] Bad DESFire GET_KEY_SETTINGS response
477836 [W][MifareDESFire] Bad DESFire GET_APPLICATION_IDS response
477860 [T][StorageAPI] File 000118F0 - 0001197C closed
477862 [T][StorageAPI] File/Dir 000118F0 free
477864 [I][NfcWorker] Unknown card. Save UID
```
Can I do anything else to help debugging?

skotopes, 2022-12-01:
I think this was fixed couple releases ago. Please confirm.

apollo13, 2022-12-01:
Hi, I am on 0.71.1 and still have problems. I will try 0.72 and report back.

apollo13, 2022-12-01:
@skotopes I just tested against 0.72 and I cannot emulate test.nfc after reading a "MIFARE DESFire EV1 (MF3ICDH41)" chip. After a reboot I can emulate test.nfc just fine again. and I cannot emulate the newly read chip either.

skotopes, 2022-12-01:
@Astrrra @gornekich can you include it in this sprint?

apollo13, 2022-12-01:
The chip that I read out looks like this:
```
Filetype: Flipper NFC device
Version: 2
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic, Bank card
Device type: UID
# UID, ATQA and SAK are common for all formats
UID: XX XX XX XX XX XX (replaced by me)
ATQA: 44 03
SAK: 20
```

Astrrra, 2022-12-08:
Fixed in #2098, please create a new issue if the problem persists.

closed 2022-12-08 by Astrrra