flipperdevices/flipperzero-firmware #1771: Weak re-played signal level on the first press

dimchansky, 2022-09-23:
### Describe the bug.

I recently tested how Flipper transmits recorded signals. To do this, I took the generated signal cleared of noise ([433.92MHz_AM650_Better_Tesla_Charge_Port_Opener.sub](https://github.com/UberGuidoZ/Flipper/blob/9a43614aabcabb8c2e73c06f2f1b654f72378768/Sub-GHz/Vehicles/Tesla/BEST_PORT_OPENER/433.92MHz_AM650_Better_Tesla_Charge_Port_Opener.sub)), which opens the Tesla charging port. I noticed that the first press of the Send button does not open the charging port, but if you press the Send button and hold it, then the port eventually opens.
At first I thought that the synthesized signal was not long enough. But later I took HackRF One to check what signal Flipper is transmitting. I noticed that the first press of the Send button transmits either a very weak signal and, perhaps, it is still transmitted with some kind of interference. But if you press the Send button and hold it down, then after the first weak signal transmission, a strong and stable signal is transmitted. You can see it on the [recorded video](https://www.youtube.com/watch?v=yFsVpPahEa4). Later, I noticed that sometimes a strong signal is transmitted from the first press of the Send button, but this rarely happens.

Is it just me or other Flipper owners have the same issue? Is this a software or a hardware issue?

My Device Info:
```
12.F7B9C6 R01:LT 0.66.1[06-09-2022] 55b46324[11528] 1.13.3:L[7] 0.66.1
```

### Reproduction

1. Send recorded signal [433.92MHz_AM650_Better_Tesla_Charge_Port_Opener.sub](https://github.com/UberGuidoZ/Flipper/blob/9a43614aabcabb8c2e73c06f2f1b654f72378768/Sub-GHz/Vehicles/Tesla/BEST_PORT_OPENER/433.92MHz_AM650_Better_Tesla_Charge_Port_Opener.sub) by pressing the Send button once.
2. Check transmitted signal level. It will be very weak.
3. Send recorded signal [433.92MHz_AM650_Better_Tesla_Charge_Port_Opener.sub](https://github.com/UberGuidoZ/Flipper/blob/9a43614aabcabb8c2e73c06f2f1b654f72378768/Sub-GHz/Vehicles/Tesla/BEST_PORT_OPENER/433.92MHz_AM650_Better_Tesla_Charge_Port_Opener.sub) by holding the Send button for 5 seconds.
4. Check the transmitted signal level. After the first transmission, the next transmissions will be much stronger. Compare this with what is recorded on [video](https://www.youtube.com/watch?v=yFsVpPahEa4).

Skorpionm, 2022-09-23:
https://youtu.be/k748otZSlEQ
my behavior does not repeat, the signal is always strong and clear, that 1 repetition is 10. the most important thing cannot be "frequency drifting" HackRF does it work fine for you?  Is it possible to check with some other remote control?

dimchansky, 2022-09-23:
> the most important thing cannot be "frequency drifting" HackRF does it work fine for you? Is it possible to check with some other remote control?

@Skorpionm I have an RTL-SDR (RTL2832U) dongle somewhere, I'll try to measure the signal level on both devices at the same time in the evening: on HackRF and on RTL-SDR.

dimchansky, 2022-09-24:
@Skorpionm Her's another setup. The signal is simultaneously received on RTL-SDR (RTL2832U) dongle and on HackRF. As requested by your colleagues on discord, I enabled SPEC mode on HackRF. Single presses of the Send button result in a very weak distorted signal at about 435.3 MHz. And as you can see in the video, sometimes a single click on Send works correctly: a strong signal is visible at a frequency of 433.92 MHz.
That is, to summarize the situation: the Tesla charging port does not open on the first press, we also do not see a strong signal at 433.92 MHz on both devices: HackRF and RTL-SDR (RTL2832U) dongle. But at the same time we do see some weak bursts at adjacent frequencies (435.3 MHz). I tend to think it's a hardware issue.
[Video](https://www.youtube.com/watch?v=RL8JxkwG8es)

dimchansky, 2022-09-24:
[Flipper static test at 433.92 MHz](https://www.youtube.com/watch?v=p_tWS8bHBCI)

dimchansky, 2022-09-24:
[Flipper carrier test at 433.919.830 Hz, path 1 - 433 MHz](https://www.youtube.com/watch?v=B1az2MdGrX0)

skotopes, 2022-09-29:
I'll close this issues since it was RMA case.

closed 2022-09-29 by skotopes