flipperdevices/flipperzero-firmware #1775: Fan/Light Universal Remote

Greendogo, 2022-09-23:
### Description of the feature you're suggesting.

Similar to the TV Universal Remote layout, and the new [FL-2792] AC Universal Remote feature, it would be handy to have a Universal Remote layout for ceiling fans which usually have light controls, several levels for fan speed (I have two, one has 3 speeds, the other has 4) and an fan speed Off button.  Some may also support more advanced features like timers/delays and automatic scheduling.

gsurkov, 2022-10-14:
We don't currently have any remote-controlled fans in the vicinity, so it would be nice to receive some more info, including:
1. The number of buttons and their functions. Must be reasonable and reflect the common features shared across different models of fans
2. The button layout (from a fan user's perspective)
3. Sample signals obtained from a typical fan remote (with manufacturer and model info)
This info would be enough to start with a basic implementation.

ferrellw, 2022-11-19:
The two remotes below are Hunter-branded ceiling fans for sale in the US. I can provide signal samples if needed.  

[A25-TX028](https://imgur.com/a/F0ItenE) - 303.875 - ASK - https://fcc.report/FCC-ID/A25-TX028/3893668.pdf

[A25-TX045](https://imgur.com/a/YXyX4De) - 433.92 - OOK - https://fcc.report/FCC-ID/A25-TX045/4712003.pdf

FWIW I am able to Read and emulate the A25-TX028 but have been running into issues with the A25-TX045 and have an open issue #2026

skotopes, 2022-12-01:
Looks like we misunderstood issue: Universal Remotes are Infrared domain.
We currently don't have and don't plan such option for SubGhz subsystem. In theory it's possible, but not going to work well because of the subghz nature.

closed 2022-12-01 by skotopes