flipperdevices/flipperzero-firmware #1783: Support for writing applications in Rust 🦀

dcoles, 2022-09-25:
### Description of the feature you're suggesting.

As an application developer, I want to be able to write applications for the Flipper Zero in [Rust](https://www.rust-lang.org/).

Initially it's going to be easiest to support [`[#no-std]`](https://docs.rust-embedded.org/book/intro/no-std.html) applications, but the long-term goal would be to also support the [Rust standard library](https://doc.rust-lang.org/std/).

## Roadmap
- [ ] Allow linking `#[no-std]` libraries to external applications: #1781
- [ ] Low-level bindings (a.k.a. `-sys` crate) for Furi/services
- [ ] High level (a.k.a "safe") bindings for Furi/services
- [ ] [`alloc`](https://doc.rust-lang.org/alloc/) support
- [ ] Evaluate how to support the [Rust standard library](https://doc.rust-lang.org/std/)
- [ ] Developer documentation

### Anything else?

At present, I'm prototyping a number of features in my [Hello, Rust](https://github.com/dcoles/flipperzero-hello-rust) application, but it would be better to move these into their own crates. Since they will be closely tied to `flipperzero-firmware`, it would probably make sense to also host Rust support infrastructure in this repository.

skotopes, 2022-09-25:
At this point we don't plan to support rust.

But we are going to accept PRs with some conditions:
- Rust runtime and standard library is not a part of firmware, it must be in elf itself
- General improvements to elf loader is ok, as long as they don't blowup internal flash usage

dcoles, 2022-09-26:
@skotopes Hi! Thanks for the information. I wanted to make sure there was an feature request, since I'm sure there will be others interested.

Rust tends to statically link the runtime/standard library, so there's no need to add support into the firmware. Which is fine if this is just for external applications on the SD card.

Since we don't want to officially support Rust, I'll create a new GitHub project for Rust bindings to the Flipper Zero and push that to [crates.io](https://crates.io/). Developers of external applications can add this as a dependency to make it easier to call Furi and services API.

dcoles, 2022-09-26:
Created [dcoles/flipperzero-rs](https://github.com/dcoles/flipperzero-rs) and published initial [`flipperzero`](https://crates.io/crates/flipperzero) and [`flipperzero-sys`](https://crates.io/crates/flipperzero-sys) crates.

dcoles, 2023-02-27:
I think we can safely close this now that there's [flipperzero-rs/flipper-zero](https://github.com/flipperzero-rs/flipperzero).

While [`std`](https://doc.rust-lang.org/std/) support is probably possible, I'm not sure it's worth the trouble at present (the Flipper Zero SDK is distinct enough from a Unix-style platform that it would require a full port).

closed 2023-02-27 by dcoles

skotopes, 2023-02-27:
@dcoles Good job guys, really awesome to see such achievements.

dcoles, 2023-02-27:
@skotopes Thanks! The SDK support has certainly made life a lot easier. The main challenge now is making the Clang-based [`bindgen`](https://crates.io/crates/bindgen) play nicely with the toolchain. Once that's reasonably stable, then we can make the [`flipperzero-sys`](https://crates.io/crates/flipperzero-sys) crate support multiple SDK versions so folks don't have to wait for me to regenerate the bindings.