flipperdevices/flipperzero-firmware #1787: Sane keyboard input not working in GDB when using the fbt tool

Semptum, 2022-09-26:
### Describe the bug.

When using `./fbt blackmagic`, a `gdb` interface appears. However, pressing backspace graphically adds a space to the prompt (as long as there is text to erase). Functionnally the text is erased but not graphically. 
Similar behavior arises as soon as the text is changed in any way such as pressing the left arrow and typing something as illustrated below.

[image]

### Reproduction

1. `cd` to the directory of the firmware
2. Turn on `gdb`
3. Type _abcd_
4. Press Left
5. Press backspace
6. Hopefully _abd_ will remain
7. Exit gdb
8. Execute `./fbt blackmagic`
9. Type _abcd_
10. Press Left
11. Press backspace
12. The result is _abcdd_ (with a space at the end)

### Anything else?

The system is Arch Linux with KDE. The Terminal emulator is Konsole but the problem persists in the virtual console and in XTerm. The shell is ZSH but the problem persists in SH and BASH.

I was advised to add "TERM" to the environment variables at line 15 of `environ.scons` but this did not help.

hedger, 2022-09-26:
@drunkbatya we might need your help with that one.

Semptum, 2022-09-26:
OK, in case it helps, it's basically an issue with ncurses. If I remove the ncurses library from the path that it creates, it works fine with the system ncurses.

skotopes, 2022-12-01:
@Semptum we've pushed couple updates, is your issue still there?

Semptum, 2022-12-01:
> @Semptum we've pushed couple updates, is your issue still there?

Yup, just checked, the problem persists on the latest dev even though it pulled in newer versions of the tools

skotopes, 2022-12-01:
@hedger @drunkbatya your turn ;)

Semptum, 2023-03-13:
> @Semptum we've pushed couple toolchain updates, do you still have this issue?

Yup, still there...

drunkbatya, 2023-03-13:
> > @Semptum we've pushed couple toolchain updates, do you still have this issue?
> 
> Yup, still there...

 Ok, I'll look at this soon

skotopes, 2023-06-22:
@drunkbatya any updates?

drunkbatya, 2023-07-01:
@Semptum 
Hey! I found a core of the problem, it was a missing libtinfo ncurses compiling dependency. I'm released a new toolchain with this fix. Can you test it?
```
git pull
git checkout drunkbatya/23-toolchain
./fbt blackmagic
```