flipperdevices/flipperzero-firmware #1797: PIN got reset with latest firmware update 0.67.2

SherifNagy, 2022-09-28:
### Describe the bug.

Once updating to the latest firmware 0.67.2, the locking PIN got reset

### Reproduction

1. Unlock flipper
2. Update firmware using qFlipper to 0.67.2
3. Check dummy mode
4. Try to lock with PIN, it will ask you to re-setup the PIN

GMMan, 2022-10-07:
This is caused by desktop settings changing version, and saved structs have no migrations. Perhaps PINs could be saved separately, preferably somewhere that would persist across firmware updates.

SherifNagy, 2022-10-07:
> This is caused by desktop settings changing version, and saved structs have no migrations. Perhaps PINs could be saved separately, preferably somewhere that would persist across firmware updates.

Make sense, this only happened once, since day one and I have PIN and did few upgrades " including one this week" to latest and didn't lose the PIN, just this one time!

skotopes, 2023-03-13:
should be fixed now

closed 2023-03-13 by skotopes