flipperdevices/flipperzero-firmware #1798: Need help with FBT error.

dtuando, 2022-09-28:
raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', 'rev-parse', '--short', 'HEAD']' returned non-zero exit status 128.
scons: *** [build/f7-firmware-D/toolbox/version.inc.h] Error 1
********** ERRORS **********
Failed building build/f7-firmware-D/toolbox/version.inc.h: Error 1

using the ./fbt command this error gets thrown. 
anyone know what this could possibly be?

hedger, 2022-09-28:
How did you get the source code? Did you download .zip archive or ran `git clone --recursive ...`?

skotopes, 2022-10-14:
Feel free to reopen issue if you still have this problem.

closed 2022-10-14 by skotopes

jayashree505, 2022-10-19:
Hi I get the same error, May I know how to overcome this?

JustDr00py, 2022-10-19:
It happened to be an issue with one of the folders. I removed it and fixed the issue

jayashree505, 2022-10-19:
Could you please tell me which folder it was?