flipperdevices/flipperzero-firmware #1804: Compact menu, that can be switch on/off in the settings.

I3elphegor, 2022-09-29:
### Description of the feature you're suggesting.

It would be great to have a compact main menu, because all icons are easy to remember and in a compact menu, you wouldn't just use the up and down buttons (much less stress on those two), but all 4 of them. Being able to turn this on and off in the settings would also be great.
[image]

skotopes, 2022-10-14:
That will require another set of icons, clearly something that we'd like to avoid. We may return to this topic in future, but clearly not till version 1.0.

skotopes, 2022-12-01:
Not going to implement: too difficult to maintain

closed 2022-12-01 by skotopes