flipperdevices/flipperzero-firmware #1806: FAP: Stable `api_version` across firmware releases

twitchyliquid64, 2022-09-30:
### Describe the enhancement you're suggesting.

As a plugin author, it would be great if I could compile an app targeting a specific API version, and then it would run on all flipper firmwares supporting it.

Ive started figuring out how to compile plugins separately to the firmware source [here](https://github.com/twitchyliquid64/fz-rust), but it would be great if it didnt need to target a specific release each time.

Basically I'm looking for somerthing similar to Androids 'min api version'.

DrZlo13, 2022-09-30:
Hello. This is how it works right now. Compiler embeds the major and minor versions of current API into elf, and major version will be checked at the loading stage of the elf loader.

Major and minor API versions do not refer to firmware versions.

DrZlo13, 2022-09-30:
But expect the API versions to increase frequently until we release version 1.0.

DrZlo13, 2022-09-30:
You can learn more about [API versioning here](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/AppsOnSDCard.md#api-versioning).

closed 2022-09-30 by DrZlo13