flipperdevices/flipperzero-firmware #1818: The common 433Mhz in Taiwan(Region3) are banned.

griffonlee, 2022-10-03:
### Describe the bug.

I got the issue when try to use 433Mhz in Taiwan(Region 3), can someone help to check?
My car's key is the 433Mhz, so I think this frqquency should not fall within the limits.

### Reproduction

1. Read/save the 433Mhz signal from Sub-Ghz page.
2. Send the saved signal.
3. Result: Transmission is blocked.

bkimminich, 2022-10-05:
It's definetely not a bug but works as designed & documented, at least with the currently available three regions:

> **R03** — The rest of the world. 
> Allowed to transmit: 312.00 - 315.25 MHz; 920.50 - 923.50 MHz.

Source: https://docs.flipperzero.one/sub-ghz/frequencies

griffonlee, 2022-10-06:
But in fact, there is no restriction on equipment with radio wave transmitting power less than 1mW by Taiwan telecommunications authority(NCC). Sincerely hope this function can be improved, thank you.

I belive flipperzero should belongs to wireless remote controls and under 1mW, so it won't break any laws in Taiwan.

## The original decree explanation is in Chinese, I only translated the part about low power devices.
https://ncclaw.ncc.gov.tw/FINT/FINTQRY04.aspx?datatype=etype&N2=10950047950&cnt=1&now=1&lnabndn=1&mode=ALL&btype=B005&recordno=1

          (7) low power radio frequency equipment, but does not include the following equipment.
                1. Passive electronic labeling of radio frequency identification equipment.
                2. Wireless chargers that do not have wireless communication functions.
                3. A wireless remote control that meets the technical specifications for low-power RF devices and has a 
                   maximum RF output power of 1 milliWatt(mW) or less.

skotopes, 2022-10-14:
I'll deploy update on this weekend

skotopes, 2022-10-16:
Sorry for delay. I've updated region database. Please update your flipper to provision new data.

closed 2022-10-16 by skotopes

griffonlee, 2022-10-18:
Thanks team!