flipperdevices/flipperzero-firmware #1821: Can't write 26-bit AWID

ae5au, 2022-10-03:
### Describe the bug.

Unable to write 26-bit AWID to T5577 tag. Able to write 36-bit AWID and other formats to same tag.

26-bit AWID will read and emulate.

### Reproduction

1. Read 26-bit AWID tag and save
2. Choose saved tag and select "Write"
3. Flipper display "Writing" and flashed purple
4. Eventually receive the "Still trying to write..." message
5. Write never completes

### Logs

```Text
Happy to gather some logs later but will need to make a non-production demo tag on Proxmark first.
```

DrZlo13, 2022-10-05:
Hello @ae5au.
Can you check #1828, does this fix this problem?

ae5au, 2022-10-05:
Hi @DrZlo13 
I don't have a dev environment handy today, but I did install 1424878d from Eng1n33r/flipperzero-firmware last night before you made this comment and it does work and includes the commits in #1828 

That's probably a good indication that the PR fixes the issue, but I'll try to build tonight and test.

ae5au, 2022-10-06:
I've confirmed that 711b5fd4 / #1828 does resolve this issue.

closed 2022-10-06 by ae5au