flipperdevices/flipperzero-firmware #1831: Big/Long Home Animations get reloaded even while scrolling through Menus

EpicLPer, 2022-10-05:
### Describe the bug.

I've noticed that having many/long/big animations on the SD Card tend to not only freeze the Flipper (which isn't the main reason for this issue :) ) but mainly scrolling through something like the Settings menu or in general some applications where the Home animation shouldn't even be loaded in or being accessed.  

I think the Flipper tries to access or read the animation data in the background even tho it's currently not even needed, and/or changes the animation that's currently playing while scrolling or clicking through menus despite the user never seeing those, generating unnecessary I/O and potential lag.

### Reproduction

1. Upload some big animations into /ext/dolphin
2. Open "Settings" and scroll/click through a few menus
3. Flipper randomly starts to freeze even tho Home Animations aren't even displayed

DrZlo13, 2022-10-07:
@EpicLPer Do you happen to have the TRACE logging level set?

EpicLPer, 2022-10-07:
> @EpicLPer Do you happen to have the TRACE logging level set?

I sadly don't and have also meanwhile moved additional animations away from the dolphin folder to prevent this, and also don't have much time currently to debug this further excuse :(  
But you can try to replicate it by extracting all the animation folders and metadata from https://github.com/RogueMaster/flipperzero-firmware-wPlugins/releases/download/0.68.2-1004-RM/0.68.2-1004-RM.tgz (go to "resources.tar" > "dolphin" and use all of those animations from there) which should make it lag then when scrolling settings menus and replicate the issue.

skotopes, 2022-10-14:
We never designed dolphin animation system to handle big sets. Nor do we plan to do that. But PR are welcomed. Meanwhile I'll close this issue.

closed 2022-10-14 by skotopes