flipperdevices/flipperzero-firmware #1851: Support R_ARM_PREL31 relocation

dcoles, 2022-10-08:
### Describe the enhancement you're suggesting.

LLVM generates `R_ARM_PREL31` relocations for `.ARM.exidx.*` symbols for [section unwinding](https://github.com/ARM-software/abi-aa/blob/main/aaelf32/aaelf32.rst#special-sections). This is needed to support features like Rust's [`panic = "unwind"`](https://doc.rust-lang.org/cargo/reference/profiles.html#panic) and C++ exception unwinding.

The workaround is to set `panic = "abort"` for Rust programs and `-fno-exceptions` flag for C++, so the program will immediately abort instead of trying to unwind the stack.

This is not a critical issue now that #1850 has been merged, but it would still be nice to support eventually.

### Anything else?

I might take a look at implementing this in the future, but since abort and exceptions typically just trigger `furi_crash`, it's not particularly urgent.

hedger, 2022-10-09:
Reference implementation: https://opensource.apple.com/source/gdb/gdb-768/src/bfd/elf32-arm.c.auto.html, look for R_ARM_PREL31.

DrZlo13, 2022-10-09:
I don't think it's a thing worth wasting time on. Stack unwinding takes up a lot of valuable memory, which increases load time and makes it impossible to write large programs.

It also requires writing support libraries for stack unwinding, AFAIK.

skotopes, 2022-10-14:
Yep, clearly not for embed.

closed 2022-10-14 by skotopes