flipperdevices/flipperzero-firmware #1870: Cannot exit picopass reader

L3K0V, 2022-10-12:
### Describe the bug.

I installed the latest one firmware and play around with the new stuff. I found in apps a picopass reader which I open. Unfortunately I do not have a picopass near me and cannot exit the app. It says "Detecting picopass card" and that's it. For 3 days I'm waiting the battery to drain. :sad

The device is not responding cannot connect to it via Bluetooth.

Help!

### Reproduction

1. Go to menu
2. Navigate to apps
3. Find Picopass reader
4. Open it
5. Device became non responding, trying to click all buttons hold them, no success

### Target

0.68.1

GMMan, 2022-10-12:
Seems to be working here. If you're seeing the hourglass animation, there might be an issue loading the app. You can reset your Flipper by pressing Left + Back.

L3K0V, 2022-10-12:
Thanks @GMMan after restart the `PicoPass Reader` is responsive and can cancel the reading the card. Maybe is something related with the firmware update and the first open of `PicoPass Reader`

closed 2022-10-14 by skotopes

LowSkillDeveloper, 2022-10-29:
I confirm. After a clean installation of the firmware, when you first open "PicoPass Reader" and click "Read Card", there is nothing else you can do. Back action doesn't work. Forced reboot only. And after the reboot, when you go in the second time everything works fine, you can go back.