flipperdevices/flipperzero-firmware #1873: Flipper not using correct key B to read sectors

Alejandro12120, 2022-10-13:
### Describe the bug.

Hello, I have a Mifare Classic 1K which is:
[Super_artefact.nfc](https://github.com/flipperdevices/flipperzero-firmware/files/9779484/Super_artefact.txt)

When I try to read it, it founds 16 keys so the auth method works but it doesn't read the sector so I think that read sector method doesn't use correct key B. Also when I use the cache file the card is read successfully.

### Reproduction

1. Create a Mifare Classic 1K structure like the above one using proxmark3
2. Enable debug logs
3. Read it using Flipper
4. Check that it founds 16 keys but it doesn't read the sectors.

gzub, 2022-10-14:
There is a bug that was fixed on another firmware(not sure if I can mention it here) that was a typo in the code, when it finds a b key that works it saves the a key by accident...  I believe there is a larger set of enhancements in another pull request misnamed mfc enhancements (or something similar) that also will likely correct this and allow clearing the cache...

Alejandro12120, 2022-10-14:
> There is a bug that was fixed on another firmware(not sure if I can mention it here) that was a typo in the code, when it finds a b key that works it saves the a key by accident...  I believe there [...]

Yes I know, I tried that fix and it allowed me to read the card successfully using the cache but using key from dictionary is bugged.

gornekich, 2022-10-14:
Hello @Alejandro12120 !
Can't reproduce this bug so far? What firmware version do you have? Can you try it on dev brunch?

Alejandro12120, 2022-10-14:
> Hello @Alejandro12120 ! Can't reproduce this bug so far. What firmware version do you have? Can you try it on dev brunch?

Hello, same issue, tried with dev f06930e4

Here are my logs

<details>
<summary>Logs</summary>
<br>

```
143024 [I][MfClassicDict] Loaded dictionary with 1 keys
143033 [D][NfcWorker] Start Dictionary attack, Key Count 1
143038 [I][NfcWorker] Sector 0
143092 [D][NfcWorker] Try to auth to sector 0 with key b0b1b2b3b4b5
143146 [D][FuriHalNfc] Timeout during data exchange
143202 [D][NfcWorker] Key found
143250 [D][NfcWorker] Trying A key for sector 1, key: b0b1b2b3b4b5
143303 [D][FuriHalNfc] Timeout during data exchange
143306 [D][NfcWorker] Trying B key for sector 1, key: b0b1b2b3b4b5
143357 [D][NfcWorker] Key found
143359 [D][MfClassic] Try to read blocks with key B
143472 [D][FuriHalNfc] Timeout during data exchange
143525 [D][FuriHalNfc] Timeout during data exchange
143578 [D][FuriHalNfc] Timeout during data exchange
143580 [D][MfClassic] Read 0 blocks out of 4
143624 [D][NfcWorker] Trying A key for sector 2, key: b0b1b2b3b4b5
143678 [D][FuriHalNfc] Timeout during data exchange
143684 [D][NfcWorker] Trying B key for sector 2, key: b0b1b2b3b4b5
143742 [D][NfcWorker] Key found
143744 [D][MfClassic] Try to read blocks with key B
143854 [D][FuriHalNfc] Timeout during data exchange
143907 [D][FuriHalNfc] Timeout during data exchange
143960 [D][FuriHalNfc] Timeout during data exchange
143962 [D][MfClassic] Read 0 blocks out of 4
144006 [D][NfcWorker] Trying A key for sector 3, key: b0b1b2b3b4b5
144060 [D][FuriHalNfc] Timeout during data exchange
144066 [D][NfcWorker] Trying B key for sector 3, key: b0b1b2b3b4b5
144118 [D][NfcWorker] Key found
144120 [D][MfClassic] Try to read blocks with key B
144230 [D][FuriHalNfc] Timeout during data exchange
144283 [D][FuriHalNfc] Timeout during data exchange
144336 [D][FuriHalNfc] Timeout during data exchange
144338 [D][MfClassic] Read 0 blocks out of 4
144382 [D][NfcWorker] Trying A key for sector 4, key: b0b1b2b3b4b5
144436 [D][FuriHalNfc] Timeout during data exchange
144442 [D][NfcWorker] Trying B key for sector 4, key: b0b1b2b3b4b5
144494 [D][NfcWorker] Key found
144496 [D][MfClassic] Try to read blocks with key B
144606 [D][FuriHalNfc] Timeout during data exchange
144659 [D][FuriHalNfc] Timeout during data exchange
144713 [D][FuriHalNfc] Timeout during data exchange
144717 [D][MfClassic] Read 0 blocks out of 4
144764 [D][NfcWorker] Trying A key for sector 5, key: b0b1b2b3b4b5
144818 [D][FuriHalNfc] Timeout during data exchange
144824 [D][NfcWorker] Trying B key for sector 5, key: b0b1b2b3b4b5
144877 [D][NfcWorker] Key found
144879 [D][MfClassic] Try to read blocks with key B
144989 [D][FuriHalNfc] Timeout during data exchange
145042 [D][FuriHalNfc] Timeout during data exchange
145095 [D][FuriHalNfc] Timeout during data exchange
145097 [D][MfClassic] Read 0 blocks out of 4
145141 [D][NfcWorker] Trying A key for sector 6, key: b0b1b2b3b4b5
145195 [D][FuriHalNfc] Timeout during data exchange
145201 [D][NfcWorker] Trying B key for sector 6, key: b0b1b2b3b4b5
145253 [D][NfcWorker] Key found
145255 [D][MfClassic] Try to read blocks with key B
145365 [D][FuriHalNfc] Timeout during data exchange
145418 [D][FuriHalNfc] Timeout during data exchange
145471 [D][FuriHalNfc] Timeout during data exchange
[... 186 lines cut ...]
153185 [D][MfClassic] Try to read blocks with key B
153288 [D][FuriHalNfc] Timeout during data exchange
153341 [D][FuriHalNfc] Timeout during data exchange
153394 [D][FuriHalNfc] Timeout during data exchange
153396 [D][MfClassic] Read 0 blocks out of 4
153398 [I][NfcWorker] Sector 14
153451 [D][NfcWorker] Try to auth to sector 14 with key b0b1b2b3b4b5
153505 [D][FuriHalNfc] Timeout during data exchange
153513 [D][MfClassic] Try to read blocks with key B
153616 [D][FuriHalNfc] Timeout during data exchange
153669 [D][FuriHalNfc] Timeout during data exchange
153722 [D][FuriHalNfc] Timeout during data exchange
153724 [D][MfClassic] Read 0 blocks out of 4
153726 [I][NfcWorker] Sector 15
153779 [D][NfcWorker] Try to auth to sector 15 with key b0b1b2b3b4b5
153833 [D][FuriHalNfc] Timeout during data exchange
153841 [D][MfClassic] Try to read blocks with key B
153946 [D][FuriHalNfc] Timeout during data exchange
153999 [D][FuriHalNfc] Timeout during data exchange
154052 [D][FuriHalNfc] Timeout during data exchange
154054 [D][MfClassic] Read 0 blocks out of 4
154174 [I][MfClassicDict] Loaded dictionary with 1242 keys
154183 [D][NfcWorker] Start Dictionary attack, Key Count 1242
```
</details>

Alejandro12120, 2022-10-17:
Hi @gornekich,

After your fix, the behaviour is the same. I will attach a video and my logs after I added this debug maybe I did it wrong since I am not a C developer.

[debug.txt](https://github.com/flipperdevices/flipperzero-firmware/files/9803106/debug.txt)

https://user-images.githubusercontent.com/45500329/196244529-4d2bf493-418b-48f9-b968-c6e55fe1e0d9.mp4

<details>
<summary>Logs</summary>
<br>

```
653006 [I][NfcWorker] Search for key cache ...
653019 [I][MfClassicDict] Loaded dictionary with 1 keys
653027 [D][NfcWorker] Start Dictionary attack, Key Count 1
653033 [I][NfcWorker] Sector 0
653089 [D][NfcWorker] Try to auth to sector 0 with key b0b1b2b3b4b5
653143 [D][FuriHalNfc] Timeout during data exchange
653194 [D][NfcWorker] Key found
653247 [D][NfcWorker] Trying A key for sector 1, key: b0b1b2b3b4b5
653302 [D][FuriHalNfc] Timeout during data exchange
653308 [D][NfcWorker] Trying B key for sector 1, key: b0b1b2b3b4b5
653360 [D][NfcWorker] Key found
653362 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
653474 [D][FuriHalNfc] Timeout during data exchange
653527 [D][FuriHalNfc] Timeout during data exchange
653580 [D][FuriHalNfc] Timeout during data exchange
653582 [D][MfClassic] Read 0 blocks out of 4
653626 [D][NfcWorker] Trying A key for sector 2, key: b0b1b2b3b4b5
653681 [D][FuriHalNfc] Timeout during data exchange
653687 [D][NfcWorker] Trying B key for sector 2, key: b0b1b2b3b4b5
653739 [D][NfcWorker] Key found
653741 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
653853 [D][FuriHalNfc] Timeout during data exchange
653906 [D][FuriHalNfc] Timeout during data exchange
653959 [D][FuriHalNfc] Timeout during data exchange
653961 [D][MfClassic] Read 0 blocks out of 4
654005 [D][NfcWorker] Trying A key for sector 3, key: b0b1b2b3b4b5
654060 [D][FuriHalNfc] Timeout during data exchange
654066 [D][NfcWorker] Trying B key for sector 3, key: b0b1b2b3b4b5
654118 [D][NfcWorker] Key found
654120 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
654232 [D][FuriHalNfc] Timeout during data exchange
654286 [D][FuriHalNfc] Timeout during data exchange
654340 [D][FuriHalNfc] Timeout during data exchange
654342 [D][MfClassic] Read 0 blocks out of 4
654387 [D][NfcWorker] Trying A key for sector 4, key: b0b1b2b3b4b5
654442 [D][FuriHalNfc] Timeout during data exchange
654448 [D][NfcWorker] Trying B key for sector 4, key: b0b1b2b3b4b5
654500 [D][NfcWorker] Key found
654502 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
654614 [D][FuriHalNfc] Timeout during data exchange
654667 [D][FuriHalNfc] Timeout during data exchange
654720 [D][FuriHalNfc] Timeout during data exchange
654722 [D][MfClassic] Read 0 blocks out of 4
654766 [D][NfcWorker] Trying A key for sector 5, key: b0b1b2b3b4b5
654821 [D][FuriHalNfc] Timeout during data exchange
654827 [D][NfcWorker] Trying B key for sector 5, key: b0b1b2b3b4b5
654879 [D][NfcWorker] Key found
654881 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
654996 [D][FuriHalNfc] Timeout during data exchange
655049 [D][FuriHalNfc] Timeout during data exchange
655102 [D][FuriHalNfc] Timeout during data exchange
655104 [D][MfClassic] Read 0 blocks out of 4
655148 [D][NfcWorker] Trying A key for sector 6, key: b0b1b2b3b4b5
655202 [D][FuriHalNfc] Timeout during data exchange
655205 [D][NfcWorker] Trying B key for sector 6, key: b0b1b2b3b4b5
655266 [D][NfcWorker] Key found
655268 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
[... 124 lines cut ...]
660493 [D][NfcWorker] Try to auth to sector 5 with key b0b1b2b3b4b5
660546 [D][FuriHalNfc] Timeout during data exchange
660550 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
660659 [D][FuriHalNfc] Timeout during data exchange
660712 [D][FuriHalNfc] Timeout during data exchange
660765 [D][FuriHalNfc] Timeout during data exchange
660767 [D][MfClassic] Read 0 blocks out of 4
660769 [I][NfcWorker] Sector 6
660822 [D][NfcWorker] Try to auth to sector 6 with key b0b1b2b3b4b5
660875 [D][FuriHalNfc] Timeout during data exchange
660879 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
660985 [D][FuriHalNfc] Timeout during data exchange
661038 [D][FuriHalNfc] Timeout during data exchange
661091 [D][FuriHalNfc] Timeout during data exchange
661093 [D][MfClassic] Read 0 blocks out of 4
661096 [I][NfcWorker] Sector 7
661148 [D][NfcWorker] Try to auth to sector 7 with key b0b1b2b3b4b5
661200 [D][FuriHalNfc] Timeout during data exchange
661308 [I][MfClassicDict] Loaded dictionary with 1242 keys
661317 [D][NfcWorker] Start Dictionary attack, Key Count 1242
```
</details>

As you can see in the logs, flipper is using correct key B, so your theory may be right, but how does it explain the cache thing?
Probably, my access bits can't be read with key B and since I am waiting for my proxmark I can't check it. Unfortunately I don't have a reader for this card.

Alejandro12120, 2022-10-19:
Don't worry take all the time you need.

Here are the logs when I read the card using cache, if that helps you.

<details>
<summary>logs</summary>

```
118755 [I][NfcWorker] Mifare Classic detected
118759 [I][NfcWorker] Trying to read a supported card ...
118765 [D][Plant] Verifying sector 8
118825 [D][FuriHalNfc] Timeout during data exchange
118831 [D][Troika] Verifying sector 11
118885 [D][FuriHalNfc] Timeout during data exchange
118889 [I][NfcWorker] Search for key cache ...
118965 [I][NfcWorker] Load keys cache success. Start reading
119069 [D][FuriHalNfc] Timeout during data exchange
119122 [D][FuriHalNfc] Timeout during data exchange
119175 [D][FuriHalNfc] Timeout during data exchange
119284 [D][FuriHalNfc] Timeout during data exchange
119337 [D][FuriHalNfc] Timeout during data exchange
119390 [D][FuriHalNfc] Timeout during data exchange
119505 [D][FuriHalNfc] Timeout during data exchange
119558 [D][FuriHalNfc] Timeout during data exchange
119611 [D][FuriHalNfc] Timeout during data exchange
119716 [D][FuriHalNfc] Timeout during data exchange
119769 [D][FuriHalNfc] Timeout during data exchange
119822 [D][FuriHalNfc] Timeout during data exchange
119931 [D][FuriHalNfc] Timeout during data exchange
119984 [D][FuriHalNfc] Timeout during data exchange
120037 [D][FuriHalNfc] Timeout during data exchange
120142 [D][FuriHalNfc] Timeout during data exchange
120195 [D][FuriHalNfc] Timeout during data exchange
120248 [D][FuriHalNfc] Timeout during data exchange
120353 [D][FuriHalNfc] Timeout during data exchange
120406 [D][FuriHalNfc] Timeout during data exchange
120459 [D][FuriHalNfc] Timeout during data exchange
120568 [D][FuriHalNfc] Timeout during data exchange
120621 [D][FuriHalNfc] Timeout during data exchange
120674 [D][FuriHalNfc] Timeout during data exchange
120779 [D][FuriHalNfc] Timeout during data exchange
120832 [D][FuriHalNfc] Timeout during data exchange
120885 [D][FuriHalNfc] Timeout during data exchange
120994 [D][FuriHalNfc] Timeout during data exchange
121047 [D][FuriHalNfc] Timeout during data exchange
121100 [D][FuriHalNfc] Timeout during data exchange
121209 [D][FuriHalNfc] Timeout during data exchange
121262 [D][FuriHalNfc] Timeout during data exchange
121315 [D][FuriHalNfc] Timeout during data exchange
121420 [D][FuriHalNfc] Timeout during data exchange
121473 [D][FuriHalNfc] Timeout during data exchange
121526 [D][FuriHalNfc] Timeout during data exchange
121635 [D][FuriHalNfc] Timeout during data exchange
121688 [D][FuriHalNfc] Timeout during data exchange
121741 [D][FuriHalNfc] Timeout during data exchange
121850 [D][FuriHalNfc] Timeout during data exchange
121903 [D][FuriHalNfc] Timeout during data exchange
121956 [D][FuriHalNfc] Timeout during data exchange
122061 [D][FuriHalNfc] Timeout during data exchange
122114 [D][FuriHalNfc] Timeout during data exchange
122167 [D][FuriHalNfc] Timeout during data exchange
122276 [D][FuriHalNfc] Timeout during data exchange
122329 [D][FuriHalNfc] Timeout during data exchange
122382 [D][FuriHalNfc] Timeout during data exchange
122384 [I][NfcWorker] Read 16 sectors out of 16 total
122396 [I][MfClassicDict] Loaded dictionary with 1 keys
122405 [D][NfcWorker] Start Dictionary attack, Key Count 1
122410 [I][NfcWorker] Sector 0
122473 [D][NfcWorker] Try to auth to sector 0 with key b0b1b2b3b4b5
122527 [D][FuriHalNfc] Timeout during data exchange
122535 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
122587 [D][MfClassic] Read 4 blocks out of 4
122590 [I][NfcWorker] Sector 1
122642 [D][NfcWorker] Try to auth to sector 1 with key b0b1b2b3b4b5
122695 [D][FuriHalNfc] Timeout during data exchange
122699 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
122750 [D][MfClassic] Read 4 blocks out of 4
122753 [I][NfcWorker] Sector 2
122805 [D][NfcWorker] Try to auth to sector 2 with key b0b1b2b3b4b5
122859 [D][FuriHalNfc] Timeout during data exchange
122867 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
122918 [D][MfClassic] Read 4 blocks out of 4
122921 [I][NfcWorker] Sector 3
122973 [D][NfcWorker] Try to auth to sector 3 with key b0b1b2b3b4b5
123026 [D][FuriHalNfc] Timeout during data exchange
123030 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
123081 [D][MfClassic] Read 4 blocks out of 4
123084 [I][NfcWorker] Sector 4
123136 [D][NfcWorker] Try to auth to sector 4 with key b0b1b2b3b4b5
123188 [D][FuriHalNfc] Timeout during data exchange
123196 [D][MfClassic] Try to read blocks with key B, b0b1b2b3b4b5
123248 [D][MfClassic] Read 4 blocks out of 4
123251 [I][NfcWorker] Sector 5
123295 [D][NfcWorker] Try to auth to sector 5 with key b0b1b2b3b4b5
123348 [D][FuriHalNfc] Timeout during data exchange
123457 [I][MfClassicDict] Loaded dictionary with 1242 keys
123466 [D][NfcWorker] Start Dictionary attack, Key Count 1242
```
</details>

colinborins, 2022-11-05:
I also have the this issue occur. Luckily i decoded my fob on another software. I just didn’t have the other device any more. I had to manually configure the file for the missing sectors.

colinborins, 2022-11-05:
I also used Mfkey 32 and it would not read all. I had 28 out of 32 read.

Aldekein, 2022-11-17:
Also have a blocker problem with the Mifare Classic 1K key.
1. Using https://github.com/equipter/mfkey32v2 I've decoded 2 keys I've got from the NFC reader.
2. I've put the found keys into `nfc/assets/mf_classic_dict_user.nfc` in uppercase.
3. Did a fresh Read of the NFC key and... It sees user dictionary and tries it during the process before starting the Flipper dictionary. But no luck, still cannot read 100% of sectors and keys :(

skotopes, 2022-12-01:
I think I saw fix arrived couple releases ago, can you guys confirm that issue is fixed?

Alejandro12120, 2022-12-01:
> I think I saw fix arrived couple releases ago, can you guys confirm that issue is fixed?

Still nothing. Same issue, reading from cache flipper thinks that all sectors are read. 
Using Dev 2a6a3a1b

skotopes, 2022-12-01:
@gornekich please take a look

Alejandro12120, 2022-12-06:
> @Alejandro12120 I am working on this issue right now. I have a question: after you read sectors with cache, do you have valid data? Could you attach resulting .nfc file after reading with cache? I mean the data from [Super_artefact.nfc](https://github.com/flipperdevices/flipperzero-firmware/files/9779484/Super_artefact.txt) file is incorrect since manufacturer block is 0 array. I wonder what flipper saves after "successful" read. It would help a lot :)

For sure here it is, [Cache.txt](https://github.com/flipperdevices/flipperzero-firmware/files/10169200/Cache.txt). The manufacturer block is 0 array, so maybe that's the problem. 

> There was a bug in reading with cache I already fixed in #2098 However I can't understand why flipper can't read data with correct key B. Due to datasheet, there is only way when MF Classic card would not allow to read data blocks with key B - when key B can be read with key A. I asked you to attach file just to see if your card is configured this way.

That will be the problem, since Flipper thinks that they can read card using key B from cache but they can't. By the way, in a few weeks my proxmark should arrive. Also, I am going to try with that pull request

Alejandro12120, 2022-12-06:
@gornekich Fixed with #2098. Congratulations, here is resulting .nfc file after #2098, [2098.txt](https://github.com/flipperdevices/flipperzero-firmware/files/10169393/2098.txt)

closed 2022-12-06 by Alejandro12120