flipperdevices/flipperzero-firmware #1880: SubGhz CAME Wrong number of bits in key

np, 2022-10-16:
### Describe the bug.

Since this [commit](https://github.com/flipperdevices/flipperzero-firmware/commit/4da6eba395652de8266ecd6a2bb63012517efd85#diff-180c2193cc29b8d3416ca37795d8d245c8ded2664321fd917ce264ed9f2f8f51) I cannot emulate one remote.

Here my `File.sub`:

```
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: CAME
Bit: 18
Key: 00 00 00 00 00 0X XX XX
```

It works up to 0.63.2-RC, then I get errors.
When emulating right after "Read", I get the error stating it is not allowed to TX to this freq (which is misleading).
When emulating from the Saved, I get a parse error.

### Reproduction

1. Load the `sub` above.
2. Emulate it

### Target

flipperzero

skotopes, 2022-10-16:
There were issues with CAME in old firmwares. Can you update to latest fw and re-capture remote?
If you'll experience issues with it please attach RAW recording

np, 2022-10-16:
In all the recent fw I've tried, re-capture still produce the same issue.

Here is the RAW:

```
Filetype: Flipper SubGhz RAW File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: RAW
RAW_Data: -2736 63 -66 131 -1054 99 -100 65 -132 97 -100 395 -66 99 -132 65 -168 297 -134 65 -66 99 -66 10716 -7362 67 -3354 65 -828 67 -468 231 -98 195 -98 165 -230 129 -1388 197 -132 459 -66 1443 -7032 67 -4946 423 -164 229 -98 525 -98 97 -66 197 -196 97 -66 97 -66 295 -132 729 -134 65 -66 1517 -66 16361 -7758 65 -4446 65 -530 299 -434 133 -132 233 -166 133 -66 365 -198 99 -728 99 -166 365 -66 1163 -7850 65 -4168 163 -98 161 -164 625 -64 131 -262 329 -432 163 -164 65 -100 265 -132 99 -232 199 -100 2097 -12352 65 -566 65 -334 199 -164 67 -232 97 -134 99 -132 99 -298 97 -132 131 -98 463 -166 99 -132 2035 -132 3401 -12310 133 -132 99 -198 99 -332 165 -1556 295 -130 259 -298 99 -362 497 -100 131 -68 963 -466 7451 -12034 231 -270 65 -166 431 -196 333 -300 363 -100 133 -134 1263 -166 2753 -7042 65 -1792 99 -100 131 -4318 65 -1762 199 -858 65 -134 99 -698 263 -332 133 -960 461 -132 263 -66 565 -132 1713 -66 18679 -11692 99 -200 133 -202 197 -134 165 -132 163 -1348 65 -230 165 -630 99 -232 959 -100 593 -132 4091 -8010 65 -2060 765 -166 165 -498 365 -564 297 -98 851 -132 197 -98 327 -164 1545 -9674 165 -3104 231 -232 525 -100 197 -430 163 -132 459 -66 229 -132 1519 -198 1253 -66 21749 -6444 65 -4674 99 -166 65 -1194 329 -786 297 -132 431 -100 163 -16526 97 -6684 67 -166 131 -100 1087 -164 1479 -660 863 -266 265 -134 333 -132 22133 -4172 299 -164 99 -132 201 -100 97 -664 65 -196 163 -396 99 -98 97 -198 99 -98 65 -164 165 -428 259 -398 1525 -66 301 -100 201 -232 1459 -784 231 -476 585 -804 227 -466 589 -436 611 -788 257 -448 613 -432 629 -756 287 -748 287 -422 643 -736 291 -420 641 -406 627 -728 325 -748 287 -756 289 -426 627 -15256 349 -686 383 -330 699 -712 355 -334 683 -364 681 -720 355 -350 655 -388 679 -688 353 -708 321 -402 657 -704 353 -356 699 -344 693 -696 323 -716 353 -688 353 -362 693 -15230 347 -722 349 -330 703 -710 341 -344 679 -384 675 -712 351 -334 683 -364 683 -704 351 -724 317 -386 653 -708 361 -388 649 -382 655 -704 353 -722 319 -728 319 -396 659 -15226 347 -722 347 -374 655 -714 345 -362 675 -384 667 -706 343 -374 681 -352 675 -708 351 -688 351 -370 681 -690 351 -386 651 -388 649 -716 353 -706 321 -732 321 -400 657 -15228 349 -720 347 -374 669 -698 323 -386 667 -374 695 -696 323 -384 687 -354 675 -712 347 -710 325 -380 667 -706 353 -358 693 -364 657
RAW_Data: -712 349 -706 349 -686 353 -360 693 -15224 383 -690 345 -362 673 -712 343 -374 677 -350 677 -710 345 -372 685 -350 673 -708 345 -704 329 -374 685 -714 345 -362 675 -354 701 -712 309 -730 347 -704 317 -382 669 -15236 383 -688 345 -376 671 -714 317 -382 667 -372 695 -692 317 -386 669 -374 695 -696 321 -716 349 -396 657 -698 323 -386 667 -410 659 -694 323 -716 353 -722 321 -394 659 -15224 347 -722 345 -376 651 -714 331 -374 673 -386 675 -708 317 -376 673 -386 673 -716 343 -692 347 -364 697 -680 351 -370 677 -380 657 -720 353 -688 335 -722 325 -380 653 -15240 357 -702 353 -394 659 -698 349 -390 639 -406 639 -736 319 -392 639 -410 673 -706 319 -726 319 -396 639 -740 319 -394 639 -408 639 -738 319 -726 317 -728 321 -400 657 -15230 347 -726 347 -362 675 -712 351 -364 657 -364 685 -722 317 -384 653 -386 679 -718 323 -710 321 -406 653 -704 355 -392 627 -398 653 -716 331 -734 319 -712 349 -394 625 -126450 65 -132 395 -132 565 -200 331 -266 429 -68 197 -134 199 -134 231 -100 99 -466 233 -300 657 -100 227 -130 691 -100 231 -100 1389 -12776 101 -134 297 -496 165 -730 199 -100 99 -266 65 -66 463 -98 131 -132 197 -100 295 -328 15447 -6684 165 -5198 99 -132 165 -132 399 -232 333 -98 399 -66 99 -266 97 -66 65 -100 131 -66 265 -200 329 -132 8503 -7036 131 -4494 67 -400 99 -366 99 -332 165 -862 99 -264 295 -264 65 -332 165 -166 365 -100 265 -132 297 -102 1027 -66 1099 -100 3405 -12560 99 -562 63 -66 199 -132 231 -1492 165 -230 65 -68 233 -166 729 -100 2659 -15484 97 -264 229 -134 499 -66 65 -68 297 -166 263 -100 299 -66 1027 -66 199 -166 2853 -12978 67 -1860 133 -332 129 -100 131 -464 65 -268 99 -196 265 -464 163 -232 327 -334 65 -332 163 -198 395 -66 755 -66 5907 -11406 129 -792 65 -132 129 -164 129 -228 163 -266 97 -66 65 -100 97 -164 131 -100 131 -296 229 -426 231 -66 497 -100 1219 -364 9459 -10364 359 -1274 63 -592 163 -298 97 -660 99 -132 199 -166 267 -364 65 -100 231 -166 265 -462 529 -100 665 -100 97 -68 2361 -13152 65 -1114 329 -100 231 -132 131 -166 263 -132 65 -100 793 -402 333 -66 1391 -15358 163 -794 67 -98 299 -1292 233 -232 67 -166 99 -694 493 -98 27371 -12182 99 -962 65 -266 365 -132 229 -794 65 -132 493 -100 663 -100 495 -98 33065 -7510 99 -2220 131 -2240 461 -230 295 -722 65 -66 131 -296 97 -98 291 -164 197 -66 789 -98 259 -12222 97 -98 197 -130 99 -66 363
RAW_Data: -198 131 -164 493 -66 99 -98 361 -298 4579 -8378 65 -4408 163 -328 97 -658 297 -230 361 -64 327 -132 493 -64 2985 -100 7891 -3782 99 -898 67 -596 165 -368 99 -860 67 -728 165 -1722 133 -234 65 -198 227 -130 261 -164 165 -100 99 -100 2791 -130 99 -66 97 -296 627 -5348 133 -4082 99 -132 99 -1194 97 -698 369 -132 329 -132 231 -98 857 -98 5071 -9326 229 -166 397 -98 165 -68 361 -166 65 -166 195 -132 195 -296 65 -166 261 -166 721 -66 1053 -100 4493 -6352 99 -98 99 -164 99 -2122 297 -636 67 -132 231 -902 531 -498 197 -134 231 -68 2015 -66 1121 -12840 163 -168 99 -332 365 -398 265 -294 391 -66 131 -132 163 -132 589 -130 195 -298 6803
```

skotopes, 2022-10-16:
Yep, I can reproduce it. @Skorpionm please take a look.

Skorpionm, 2022-10-17:
it's not came, which has either 12 or 24 bits key ... attach a photo of the remote control

np, 2022-10-18:
Here it is:
[image]

Skorpionm, 2022-10-18:
Check https://github.com/flipperdevices/flipperzero-firmware/pull/1890

closed 2022-10-18 by skotopes

np, 2022-10-19:
Thanks you rock guys!