flipperdevices/flipperzero-firmware #1886: NFC st25ta support

SkalkaA, 2022-10-17:
### Description of the feature you're suggesting.

Bought a smart lock ROTHULT made by IKEA, which uses st25ta 14443A cards for access. 
Currently it is not possible to emulate these cards with a Flipper.  

It is possible to read/emulate them, and to extract the password using a Proxmark3 in standalone mode by touching the lock itself, thereby making a copy of the master card*. It would be great if something similar could be done on the Flipper.

*This shows the cloning mentioned: [https://www.youtube.com/watch?v=Q08qhJ3TOM8&ab_channel=QuentynTaylor](url)

woodyc79, 2023-02-22:
Interesting: If i go to NFC/Detect Reader, the IKEA lock will open/close!

zeano, 2023-02-22:
> 

Hello, i noticed the same thing using Detect Reader on my IKEA lock.

MichaelGrafnetter, 2023-06-13:
> Interesting: If i go to NFC/Detect Reader, the IKEA lock will open/close!

@woodyc79 @zeano  This is expected behavior, as in the unlocked state, the ROTHULT device can be locked with any other card and then re-opened with itsame card (same UID). And the Detect Reader feature of Flipper just emulates a Mifare card with a constant UID.

ChrisD0lpgr3n, 2023-10-10:
I have the same problem: Emulating cards are not recognized 
But that's not a main problem with the lock: a tag that I had previously cloned to a Magic Tag, worked 

I'll test whether it works if I clone the card/chip read with Flipper onto a Magic Tag using the Zero

ChrisD0lpgr3n, 2023-10-10:
> I have the same problem: Emulating cards are not recognized But that's not a main problem with the lock: a tag that I had previously cloned (RFID Tools and PN532) to a Magic Tag, worked
> 
> I'll test whether it works if I clone the card/chip read with Flipper onto a Magic Tag using the Zero

Now the chip, read out with a flipper, is written onto a magic tag, which Rothult opens and closes
Emulating the same chip doesn't work

(I can't test with the original Ikea cards, the Magic Tag doesn't accept them)

skotopes, 2024-01-03:
Looks like this issues already solved.

skotopes, 2024-01-04:
@bettse true, we'll take a look into st24ta implementation. However we still lack cards and systems that uses it. 

@doomwastaken can you take a look on it, we need cards and this lock.

bettse, 2024-01-04:
Are their IKEA anywhere nearby?  https://www.ikea.com/us/en/p/rothult-smart-lock-white-00429619/

skotopes, 2024-01-04:
Surprisingly not. Ikea japan is not selling this lock.

doomwastaken, 2024-01-04:
I will order them to one of the offices

doomwastaken, 2024-01-04:
Ordered it, will have in few hours

SkalkaA, 2024-01-19:
I do have a Proxmark3, IKEA lock and the cards. Flipper also, of course. If you need help beta testing, I definitely could try.

afroewis, 2024-06-22:
I bought a Rothult and a Flipper Zero and confirm that Flipper is not able to unlock the Rothult by emulating the shipped card. In my case, it also doesn't work one minute within opening the lock with the shipped card. 

Would be awesome if you could implement this at some point. Anyway, thanks for all your work!

skotopes, 2024-06-22:
@SkalkaA @afroewis please ping @gornekich