flipperdevices/flipperzero-firmware #1895: Flipper just keeps on painting

joooon, 2022-10-19:
### Describe the bug.

Hello friends,

I've recently upgraded to fw 0.69.0 and immediately noticed that my dolphin has picked up a new hobby and am very pleased with his new occupation.

However I've also noticed that he seems to be doing little else and keeps on painting in every other animation. Don't get me wrong, I'm happy he's not destroying his eyes in front of that CRT or breathing solder fumes all the time - but I wouldn't want him to lose his other skills.

Jokes aside, I see that the painting animation [has a weight of 6](https://github.com/flipperdevices/flipperzero-firmware/blob/9a9abd59e9a97e0e215041eca7fd1ee0570bd10b/assets/resources/dolphin/manifest.txt#L93), while the rest of them use a value of 3.
I wasn't able to find the reason for this in the commit and pull request, so I actually wonder if this is expected.

In any case I suggest we move the weight of the animation in line with the rest. I'd be happy to test the change and submit a pull request.

### Reproduction

1. Switch on
2. Stare at screen

UberGuidoZ, 2022-10-21:
😂

skotopes, 2022-10-22:
In terms of probability and math it behaves as intended.
But we are going to change balance in future releases. This is seasonal thing ;-)

closed 2022-10-22 by skotopes