flipperdevices/flipperzero-firmware #1896: build fails when using ./fbt

jayashree505, 2022-10-19:
### Describe the bug.

subprocess.CalledProcessError: Command '['git', 'rev-parse', '--short', 'HEAD']' returned non-zero exit status 128.
        CC      lib/ST25RFAL002/source/st25r3916/st25r3916_led.c
        CC      lib/ST25RFAL002/source/custom_analog_config.c
scons: *** [build/f7-firmware-D/toolbox/version.inc.h] Error 1
********** ERRORS **********
Failed building build/f7-firmware-D/toolbox/version.inc.h: Error 1

### Reproduction

runing the ./fbt

hedger, 2022-10-19:
How did you get the source code? GIt clone or zip archive?

jayashree505, 2022-10-19:
I git cloned it using the following command:

`$ git clone --recursive https://github.com/flipperdevices/flipperzero-firmware.git`

hedger, 2022-10-20:
Can you show the output of `git rev-parse --short HEAD` in repo root dir?
And the return code after running it manually, `echo $?`

jayashree505, 2022-10-20:
On running `git rev-parse --short HEAD`, I get `9a9abd59`

May I know which return code you are looking for?

hedger, 2022-10-20:
I'm looking for that command's return code. It can be displayed after running it with `echo $?`.
For sone reason, that command fails when it's being invoked by fbt. Let's make sure its return code is 0 when running in normal shell.

After that, please spawn fbt shell (by running `source scripts/toolchain/fbtenv.sh`) and re-run git command and return code check.

jayashree505, 2022-10-20:
Hi, On running git rev-parse --short HEAD, I get 9a9abd59 in both cases. Also, the return code is 0 in both cases.

pavllan90, 2022-11-22:
Same problem. Return code is 0 in both cases.

skotopes, 2022-12-01:
On which platform you guys trying to build?

steam3d, 2022-12-11:
I have the same issue. I use VirtualBox with Ubuntu.

>Distributor ID:	Ubuntu
Description:	Ubuntu 22.10
Release:	22.10
Codename:	kinetic

`git rev-parse --short HEAD`

Result:
```
27921e42
```

`source scripts/toolchain/fbtenv.sh`

Result:
```
Please source this script from [u]fbt root directory, or specify 'SCRIPT_PATH' variable manually
Example:
	SCRIPT_PATH=lang/c/flipperzero-firmware source lang/c/flipperzero-firmware/scripts/fbtenv.sh
If current directory is right, type 'unset SCRIPT_PATH' and try again
```

hedger, 2023-01-11:
Tried multiple system configurations,  couldn't reproduce. Maybe it's caused by incomplete/corrupted repo clone. 
Closing for now, reopen if you can provide additional logs or debugging info.

closed 2023-01-11 by hedger