flipperdevices/flipperzero-firmware #1902: Run bad USB script over bluetooth

krisss117, 2022-10-21:
### Description of the feature you're suggesting.

Hi, plese add function to send BAD USB script over Bluetooth. 

1. Connect Flipper Keyboard over Bluetooth
2. Allow to select BAD USB script to run

skotopes, 2022-10-22:
Will require RPC mode implementation in BadUSB application and corresponding part in apps. We'll come back to this task after application catalog.

ggaljoen, 2022-12-26:
Running scripts will also require to read, and if needed toggle, the keyboard (caps/num) locks status, otherwise you won't get the intended key strokes.

UnsavorySpirit, 2023-04-10:
> 

if you install the xtreme fw (i use unleashed) you can see te badkb function and now i'm trying to transfer that function to the unleashed fw is there a way you could do thath or at least help me?

krisss117, 2023-05-04:
Finctionality was add in Xtreme Firmware :)

UnsavorySpirit, 2023-05-04:
I already know

intech, 2023-07-31:
@skotopes The application catalog is already running. Can we expect updates on this task shortly?

skotopes, 2023-08-23:
@intech sorry for delay, we'll try to pack it in 1.0 release

sk337, 2023-11-11:
I would like to see this feature as well but it seems like this has been forgotten

skotopes, 2024-01-03:
@sk337 more like it's coming as a part of the scripting language

skotopes, 2024-07-07:
Done

closed 2024-07-07 by skotopes