flipperdevices/flipperzero-firmware #1904: NFC 0.69.1 Can not parse file

Antiarchitect, 2022-10-21:
### Describe the bug.

I have two phones with NFC chips. Reading one (One Plus 7) goes normally and I can Emulate UID normally. But have issues with another one which is Honor 50: I can read NFC and save it but selecting saved file tells `Can not parse file` and thus I have no actions menu (UX bug) so I cannot even delete it.
[image]
[image]

### Reproduction

1. Go to NFC
2. Select Read
3. Read Honor 50 with NFC turned on
4. Save and name
5. Try to select the file in Saved section

gornekich, 2022-10-21:
Hello @Antiarchitect !
Could you please share the content of the saved file? Nfc type and UID fields in particular

gornekich, 2022-10-21:
Not UID, but it's length

Antiarchitect, 2022-10-21:
Will try to extract the file a bit later :)

Antiarchitect, 2022-10-21:
[https://dev.flpr.app/s#path=nfc/Vika.nfc&Filetype=Flipper+NFC+device&Version=2&Device+type=UID&UID=00+00+00+00+00+00+00+00+00+00&ATQA=00+00&SAK=00](url)

Antiarchitect, 2022-10-21:
https://dev.flpr.app/s#path=nfc/Vika.nfc&Filetype=Flipper+NFC+device&Version=2&Device+type=UID&UID=00+00+00+00+00+00+00+00+00+00&ATQA=00+00&SAK=00

LowSkillDeveloper, 2022-10-29:
I have the same problem with my bus pass.

```
Filetype: Flipper NFC device
Version: 2
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic, Bank card
Device type: Mifare DESFire
# UID, ATQA and SAK are common for all formats
UID: 04 73 43 85 97 7D 80
ATQA: 44 03
SAK: 20
# Mifare DESFire specific data
PICC Version: 04 01 01 01 00 1A 05 04 01 01 01 04 1A 05 04 73 43 85 97 7D 80 B9 0C 22 41 50 05 22
PICC Free Memory: 1312
PICC Change Key ID: 00
PICC Config Changeable: true
PICC Free Create Delete: false
PICC Free Directory List: true
PICC Key Changeable: true
PICC Max Keys: 01
PICC Key 0 Version: 00
Application Count: 10
Application IDs: D0 02 F4 70 02 F0 60 34 F5 10 20 F1 50 89 F8 20 20 F1 80 10 F1 A0 11 F1 0B 10 00 FF FF FF
```

[image]

mh-, 2022-11-22:
> I have the same problem with my bus pass.

Ok, yes that's the same problem as in #2040 - the DESFire import 
https://github.com/flipperdevices/flipperzero-firmware/blob/820afd2aec377629c3e514ceb57ecdd0f9163b6c/lib/nfc/nfc_device.c#L426
expects details about the key for each application.

If you modify the file like this
```
Filetype: Flipper NFC device
Version: 2
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic, Bank card
Device type: Mifare DESFire
# UID, ATQA and SAK are common for all formats
UID: 04 73 43 85 97 7D 80
ATQA: 44 03
SAK: 20
# Mifare DESFire specific data
PICC Version: 04 01 01 01 00 1A 05 04 01 01 01 04 1A 05 04 73 43 85 97 7D 80 B9 0C 22 41 50 05 22
PICC Free Memory: 1312
PICC Change Key ID: 00
PICC Config Changeable: true
PICC Free Create Delete: false
PICC Free Directory List: true
PICC Key Changeable: true
PICC Max Keys: 01
PICC Key 0 Version: 00
Application Count: 0
```
i.e. set Application Count to 0, Flipper can read the file and emulate the UID.

But: I would advise against trying that UID in an actual bus, assuming that you are not the owner of the bus company or authorized to do penetration testing by them. And also, it is unlikely that such a system will accept the clone just based on the UID.

skotopes, 2022-11-28:
Fixed in #2041 thanks to @mh-