flipperdevices/flipperzero-firmware #1909: App/SubGhz proto: support TPMS (Tire Pressure Monitoring System) ?

u1735067, 2022-10-22:
### Description of the feature you're suggesting.

I don't know if it is relevant, but since I couldn't find any mention of TPMS already, maybe an app to decode TPMS (Tire Pressure Monitoring System) messages could be implemented, and maybe even with a "Relearn" option (https://github.com/UberGuidoZ/Flipper/tree/main/Sub-GHz/Vehicles/EL50448_-TPMS_Relearn_Tool)?

### Anything else?

This is one of the things https://github.com/merbanan/rtl_433 supports, to have an implementation example.

skotopes, 2022-10-22:
Should be totally feasible. All APIs are available and can be implemented as external application.
At the same time it's kinda niche thing, we don't plan to implement it. Maybe someone from community will.

Crsarmv7l, 2022-10-31:
@skotopes Any plans to implement FSK? It would be needed for TPMS, afaik flipper only supports 2FSK

skotopes, 2022-10-31:
2fsk is fsk. All modulations supported by CC1101 are available to user applications.

Crsarmv7l, 2022-10-31:
Hm. Well I guess I need to play around more with my deviation.

Crsarmv7l, 2022-11-05:
Got it. 3 out of 4 test captures are recognized as TPMS by rtl_433. Only implemented the modulation for Toyota ATM, and haven't even touched writing a decoder

Crsarmv7l, 2022-11-06:
@u1735067 

If anyone cares here is the modulation for Toyota PMV_107J TPMS
```
Custom_preset_name: TPMS
Custom_preset_module: CC1101
Custom_preset_data: 02 0D 07 04 08 32 0B 06 10 69 11 93 12 14 13 01 14 F8 15 40 18 18 19 16 1B 07 1C 00 1D 91 20 FB 21 56 22 10 00 00 C0 00 00 00 00 00 00 00 
```
Using a custom fw is recommended as you will need extended freqs to utilize 314.98Mhz. Detect Raw is also needed because there is no decoder for TPMS.

I have run multiple tests (about 40ish captures in total) with this modulation/freq and every signal captured except one was the targeted Toyota TPMS. This was tested by replaying the captured and save Raws to rtl_433.

u1735067, 2023-07-31:
While I would love having the time to play with this (I also don't have TPMS sensors), I saw other peoples made progress, so to share:
* https://github.com/wosk/flipperzero-tpms
* https://github.com/Crsarmv7l/TPMS-Flipper

Crsarmv7l, 2023-07-31:
Your second link is mine ;). That said I had seen the first one recently. I
may have to take a look and see if I can modify it to tx

u1735067, 2023-07-31:
Yep, I though it might be useful to have it here too :)

And by playing with new apps, I just found that the ProtoView app (https://github.com/antirez/protoview) also implements TPMS: https://github.com/antirez/protoview/tree/main/protocols/tpms 

<sub>Btw I'm surprised to see the repo associated in the catalog at https://github.com/flipperdevices/flipper-application-catalog/blob/main/applications/Sub-GHz/protoview/manifest.yml, it makes look like many app come from the same repo, while they're not actually ..</sub>

skotopes, 2024-01-03:
I think at this point we are not going to implement it as a part of firmware.

closed 2024-01-03 by skotopes