flipperdevices/flipperzero-firmware #1915: Flipper repeatedly shows internal storage full when attempting to flash firmware

d0minikkkk, 2022-10-23:
### Describe the bug.

when attempting to flash the firmware updates via `fbt updater_package`, the flipper always show unsuccessful with error message: Internal Storage is full.

### Reproduction

1. Compiler firmware
2. Upload to SD card
3. flash firmware

skotopes, 2022-10-23:
It says so for a reason. Usually it means that you have only couple pages left.
How big is your firmware, how much space is left on internal storage?

d0minikkkk, 2022-10-23:
> It says so for a reason. Usually it means that you have only couple pages left.
> 
> How big is your firmware, how much space is left on internal storage?

My internal storage is totally used. How can i clear up internal storage?

skotopes, 2022-10-23:
You can use qFlipper to manage files.

d0minikkkk, 2022-10-23:
qflipper shows several files within internal flash, but no sizes are indicated. May i find out what files are crucial and what can I remove to free up space?

skotopes, 2022-10-23:
critical files are starting with `.`

d0minikkkk, 2022-10-23:
kindly refer to my updated comment with screenshot. it seems that there isn't any other files apart from crucial files.

skotopes, 2022-10-23:
Yep, all of them are critical.
So how much free space do you have. What's in your firmware that occupies so much?

d0minikkkk, 2022-10-23:
Currently, i have 0kb free.
I am currently using the firmware in the release branch, with slight modifications to remove region locks.

skotopes, 2022-10-23:
Oh, that firmware. Please talk with that firmware maintainer. We only support official firmware here.

closed 2022-10-23 by skotopes

d0minikkkk, 2022-10-23:
I believe there is a misunderstanding. I am using the firmware here, but just with the region checks removed.

skotopes, 2022-10-23:
Picopass is included in official firmware as external application and should not be compiled into firmware.
Open "Menu->Setting->Storage->About internal storage": you should see 116kb there.
In case if you see 116k and 0k free it most likely lfs corrupt. Use "Factory Reset" to restore it.

d0minikkkk, 2022-10-23:
> Picopass is included in official firmware as external application and should not be compiled into firmware. Open "Menu->Setting->Storage->About internal storage": you should see 116kb there. In case if you see 116k and 0k free most likely lfs is corrupted. Use "Factory Reset" to restore it.

I only see 24KB total, 8KB free on my device. Kindly advice steps going forward

skotopes, 2022-10-23:
8kb is 2 pages. That is what updater is telling to you.

qFlipper can install dfu files directly, but internal storage in that case may be wiped. Same goes for flashing in recovery mode: it always succeed but internal data will be lost.

Don't compile into firmware apps compile them as external apps.

d0minikkkk, 2022-10-23:
> 8kb is 2 pages. That is what updater is telling to you.
> 
> qFlipper can install dfu files directly, but internal storage in that case may be wiped. Same goes for flashing in recovery mode: it always succeed but internal data will be lost.
> 
> Don't compile apps into firmware, compile them as external apps.

How can I flash new firmware onto the device in the current state? I am unable to flash any firmware due to the space limitation. Even booting into dfu mode and repairing it is not working.

skotopes, 2022-10-23:
qFlipper is always can do flash from recovery:

- Press left + back 
- Release left - device will go into recovery mode
- open qFlipper and press recover button