flipperdevices/flipperzero-firmware #1918: Cannot build on M1 MacBook Pro (assets_icons.h not found)

zmactep, 2022-10-24:
### Describe the bug.

I have an error that `assets_icons.h ` header is not found during the build process.

### Reproduction

```
$ git clone --recursive https://github.com/flipperdevices/flipperzero-firmware.git
$ cd flipperzero-firmware
$ ./fbt
```

### Target

firmware

### Logs

```Text
pavel@MacBook-Pro-14 [11:42:12] [~/SynologyDrive/DEV/tools/flipper/firmwares/flipperzero-firmware] [dev]
-> % ./fbt 
	APPS	build/f7-firmware-D/applications/applications.c
	ICONS	build/f7-firmware-D/assets/compiled/assets_icons.c
	CC	applications/main/archive/scenes/archive_scene.c
	CC	applications/main/bad_usb/scenes/bad_usb_scene.c
	PREGEN	build/f7-firmware-D/sdk_origin.i
	SDKSRC	build/f7-firmware-D/sdk_origin.i
	CC	applications/main/gpio/scenes/gpio_scene.c
	CC	applications/main/gpio/gpio_item.c
	CC	applications/main/gpio/usb_uart_bridge.c
	CC	applications/main/ibutton/scenes/ibutton_scene.c
	CC	applications/main/ibutton/ibutton_cli.c
	CC	applications/main/infrared/scenes/infrared_scene.c
In file included from applications/services/gui/canvas.h:9,
                 from ./applications/services/dialogs/dialogs.h:3,
                 from build/f7-firmware-D/sdk_origin.i.c:124:
applications/services/gui/icon_animation.h:10:10: fatal error: assets_icons.h: No such file or directory
   10 | #include <assets_icons.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
scons: *** [build/f7-firmware-D/sdk_origin.i] Error 1
********** ERRORS **********
Failed building build/f7-firmware-D/sdk_origin.i: Error 1
```
```

DrZlo13, 2022-10-24:
Known issue. Workaround: run the build again and it will be fixed.

zmactep, 2022-10-24:
Miracle. Works for me, thanks!

closed 2022-10-24 by zmactep