flipperdevices/flipperzero-firmware #1932: Acurite-606TX not picking up weather sensor

PolymerPrints, 2022-10-26:
### Describe the bug.

Trying to get the information off a weather sensor using the Acurite-606TX protocol with the weather station app (v0.3). Using an RTL-SDR and rtl_433, the information is correctly read and identified as Acurite-606TX @ 433.92MHz. Flipper is just stuck on "Scanning..." at the same frequency.

rtl_433 output:
[image]

### Reproduction

1. Open weather station app
2. Start reading
3. Trigger TX event on weather station
4. Nothing happens

PolymerPrints, 2022-10-26:
Raw recording can be found here:
[acurite_issue.zip](https://github.com/flipperdevices/flipperzero-firmware/files/9873646/acurite_issue.zip)

PolymerPrints, 2022-10-27:
That works great! Thanks for the quick fix!

Skorpionm, 2022-10-27:
close the issue if the error is fixed)

closed 2022-10-28 by PolymerPrints