flipperdevices/flipperzero-firmware #1934: Hopper frequency presets

LowSkillDeveloper, 2022-10-26:
### Description of the feature you're suggesting.

It would be great if you could make different Hopper_frequency presets.
And instead of the ON and OFF switch, you choose a preset.

nuschpl, 2022-11-06:
It would be also useful to keep On / Off along with this feature or rename it to FreqLock or something. I've got use case for this: I've got one remote of a car vendor which name starts with T which is simply toggling between two frequenencies - each subsequent press of the same button generates signal on different freq but you can be 100% sure if you will press twice you will get both freqs. So the best implementation of this if we would like to capture subsequent key presses would be to start with frequency hopping, then after processing initial transmission disabling it and toggle frequency programmatically to the alternative other one to stay in sync with remote's logic

varenc, 2023-06-30:
I would love this feature too, and I love the proposed implementation. In particular, I just want to hop between the narrow set of frequencies have the traffic I'm looking for, and not other frequencies when I know aren't even in use in my region.

Off-topic but, does anyone know of any 3rd party utils for the flipper that this possible?