flipperdevices/flipperzero-firmware #1967: Weather Station: Add Bresser 5-in-1 protocol

AlexeyOplachko, 2022-11-01:
### Description of the feature you're suggesting.

Would be nice to have support for Bresser 5-in-1 weather stations (and maybe their other ones as well)
It seems like in RTL_433 they already did most of the work on decoding protocol: https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_5in1.c

I have access to one of these(technically it's a different model than mentioned in the code, but it has same outdoor unit) I can do a RAW recording if you have advice on which settings should I use for recording

Skorpionm, 2022-11-03:
There will be RAW recordings, there will be their support. open SubGhz RAW and record 2+ messages from sensors. + set RSSI -85 in the settings and also make 2+ records (preferably on different channels if it is supported) it is also important to take a photo of what the receiver shows (temperature, humidity, channel) or RTL433. throw records for example here or on the forum, I will add support

AlexeyOplachko, 2022-11-03:
It's on 868mhz and if I understood terms used in RTLs decoder correctly, it uses FM? Is there a way to determine if it's 238 or 476?

Skorpionm, 2022-11-03:
Yes, you are right, I didn't look at the source code in RTL. The easiest option is to write it down on both modulations. send them to me and I'll take a look.
 if you want to check, record the file, make the temperature change, and then play the recorded one, the receiver will return to the old temperature, frequency 868.35

skotopes, 2022-12-01:
@AlexeyOplachko we need RAW recording of this protocol

AlexeyOplachko, 2022-12-02:
I've tried recording and whatever I managed to record wasn't being recognized by receiver part of the station, so no recording. And for unrelated reasons station broke so can't try to record it again for a while anyway. I guess we can close this

closed 2023-03-13 by skotopes