flipperdevices/flipperzero-firmware #1976: Missing wheel in scripts/requirements.txt

lslrt, 2022-11-05:
### Describe the bug.

Running `pip3 install -r scripts/requirements.txt` in a fresh python3 virtual env fails with the following error:
```
Building wheels for collected packages: heatshrink2
  Building wheel for heatshrink2 (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/lsa/Projects/flipperzero-firmware/scons-venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-5de3j9t1/heatshrink2/setup.py'"'"'; __file__='"'"'/tmp/pip-install-5de3j9t1/heatshrink2/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-_u9leoqe
       cwd: /tmp/pip-install-5de3j9t1/heatshrink2/
  Complete output (6 lines):
  usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: setup.py --help [cmd1 cmd2 ...]
     or: setup.py --help-commands
     or: setup.py cmd --help
  
  error: invalid command 'bdist_wheel'
  ----------------------------------------
  ERROR: Failed building wheel for heatshrink2
  Running setup.py clean for heatshrink2
Failed to build heatshrink2
```
To fix the build issue, a solution is to pip install wheel.
`pip3 install wheel`
Then run again
`pip3 install -r scripts/requirements.txt`

### Reproduction

`python3 -m venv scons-venv`
`source scons-venv/bin/activate`
`pip3 install -r scripts/requirements.txt`

### Target

Build system

skotopes, 2022-11-05:
Normally those requirements are included in toolchain.

skotopes, 2022-11-05:
Any reason why you want to use python other than included in toolchain?

lslrt, 2022-11-05:
@skotopes I'm just following https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/fbt.md, there is no mention of toolchain anywhere.
Is there is some other instructions that I've missed?

skotopes, 2022-11-05:
Yep, ignore this part. The only requirement is git and wget/curl to download toolchain. Everything else will be automatically pulled by fbt.

@hedger it make sense to clarify it in documentation.

lslrt, 2022-11-05:
@skotopes ok, thank!

closed 2022-11-05 by lslrt