flipperdevices/flipperzero-firmware #1977: Blackmagic debug not working on Linux x64

augustozanellato, 2022-11-05:
### Describe the bug.

Blackmagic debugger crashes with a python startup error, it probably affects all the debugging methods. 
It's probably caused by a toolchain issue.

### Reproduction

1. Clone the repo
2. Run `./fbt blackmagic`

### Logs

```Text
arm-none-eabi-gdb-py -ex "target extended-remote tcp:192.168.1.189:2345" -ex "set confirm off" -ex "set pagination off" -ex "monitor swdp_scan" -ex "monitor debug_bmp enable" -ex "attach 1" -ex "set mem inaccessible-by-default off" -ex "source ./debug/FreeRTOS/FreeRTOS.py" -ex "source ./debug/flipperapps.py" -ex "fap-set-debug-elf-root build/f7-firmware-D/.extapps" -ex "source ./debug/PyCortexMDebug/PyCortexMDebug.py" -ex "svd_load ./debug/STM32WB55_CM4.svd" -ex compare-sections build/f7-firmware-D/firmware.elf
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Python path configuration:
  PYTHONHOME = (not set)
  PYTHONPATH = (not set)
  program name = '/vdb/tool/temp/python/bin/python'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/vdb/tool/temp/python/bin/python'
  sys.base_prefix = '/vdb/tool/temp/python'
  sys.base_exec_prefix = '/vdb/tool/temp/python'
  sys.platlibdir = 'lib'
  sys.executable = '/vdb/tool/temp/python/bin/python'
  sys.prefix = '/vdb/tool/temp/python'
  sys.exec_prefix = '/vdb/tool/temp/python'
  sys.path = [
    '/vdb/tool/temp/python/lib/python39.zip',
    '/vdb/tool/temp/python/lib/python3.9',
    '/vdb/tool/temp/python/lib/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007f9fa21b4040 (most recent call first):
<no Python frame>
scons: *** [phony_blackmagic] Error 1

********** FBT ERRORS **********
phony_blackmagic: Error 1
```

skotopes, 2022-11-10:
Check latest dev or RC please

augustozanellato, 2022-11-10:
> Check latest dev or RC please

Working fine on latest dev, thanks!

closed 2022-11-10 by augustozanellato