flipperdevices/flipperzero-firmware #1988: Incorrect SAK in Mifare emulation and magic card writing

tkerby, 2022-11-08:
### Describe the bug.

When the flipper reads the card, it correctly reports the hardware SAK (e.g. when you select info after reading a card), but it often appears to fail to emulate NFC cards correctly, and likewise, cards produced with the magic card writing functionality also fail to work with readers.

This appears to be that when emulating or writing, the Flipper uses the Block 0 data stored in the Mifare card to generate the SAK rather than the value otherwise stored in the saved .nfc files SAK tag. This data can be intentionally incorrect to prevent cards from being copied, resulting in failures (e.g. hotel door locks). See https://gist.github.com/equipter/3022aea4e371e585ff6e46de637e7769 for an explanation

### Reproduction

Read a mifare card and check the SAK. See if the block 0 data in the saved capture matches the SAK

If the block 0 data is different (security measure being used) then try emulating a card or writing a magic card and see if they work with a reader. 

### Anything else?

To fix this, we should emulate and write the SAK to cards using the SAK that the flipper reads rather than the block 0 value. The block 0 value should only be reported on a direct read to that block

tkerby, 2022-11-08:
This seems to be related to https://github.com/flipperdevices/flipperzero-firmware/issues/1632 and might explain the behaviour and fix

equipter, 2022-11-09:
> To fix this, we should emulate and write the SAK to cards using the SAK that the flipper reads rather than the block 0 value.

that's a good idea as long as it doesn't begin to mess with other mifare classic types like the ones that *do* use sak 0x88 and so on. 

there's also the problem with some access control providers doing the sneaky thing of making sure that the sak on the initial read and the block 0 are different. paxton have a bit of history with this.

tkerby, 2022-11-09:
> that's a good idea as long as it doesn't begin to mess with other mifare classic types like the ones that _do_ use sak 0x88 and so on.
>
> there's also the problem with some access control providers [...]

I think to combat this we should try an initial read and a block 0 read and store both values. It should be possible to emulate both. I believe the flipper already does this on read as the .nfc file can show a different sak in the SAK: field to the data its read in block 0

SaleSlave, 2022-11-17:
I hope it may help

[image]
[image]

doomwastaken, 2023-11-03:
@tkerby @SaleSlave could you please verify your issue with latest firmware? We've merged NFC refactor branch

doomwastaken, 2023-11-21:
Please reopen if issue persists

closed 2023-11-21 by doomwastaken