flipperdevices/flipperzero-firmware #1996: Bluetooth not working

Emzarii, 2022-11-10:
### Describe the bug.

After I updated software my Bluetooth connection on flipper stopped switching on bt module, so that I cannot connect to any device via bt

### Reproduction

Do not know what to write here

LowSkillDeveloper, 2022-11-10:
I have the same problem after using "Bluetooth Remote".  After using it on the phone, the phone cannot connect to the flipper through the app. I solved by delete the pair in the bluetooth settings on phone and flipper and data reset of flipper application.

skotopes, 2022-12-01:
Hi guys, we need logs from flipper and your device to reproduce and fix this issue. Also device model/OS will be nice to know.

skotopes, 2022-12-20:
There is a fix in latest dev, which also will be included in next release. Please check and reopen if your issue is still persist.

closed 2022-12-20 by skotopes

Davidberzerk, 2023-12-24:
I have most current iOS and my flipper Bluetooth won't work since I put extreme on it I'll delete flipper and reinstall and see

skotopes, 2023-12-24:
@Davidberzerk we don't provide support for 3rd party firmware.