flipperdevices/flipperzero-firmware #2013: Universal Remote not working on new Sony TV’s

ALD53105, 2022-11-13:
### Describe the enhancement you're suggesting.

Universal Remote does not change volume or channels on newer Sony TV’s. Only the power can be cycled.

Crsarmv7l, 2022-11-13:
Many of the newer tv's use ir for the power button and bluetooth for the other buttons.

ALD53105, 2022-11-15:
> Many of the newer tv's use ir for the power button and bluetooth for the other buttons.

I can read and duplicate the IR signals from the remote with flipper and control both volume and channel on the TV.

Crsarmv7l, 2022-11-17:
> > Many of the newer tv's use ir for the power button and bluetooth for the other buttons.
> 
> I can read and duplicate the IR signals from the remote with flipper and control both volume and channel on the TV.

Ok so that directly contradicts the issue you opened.

ALD53105, 2022-11-18:
> Ok so maybe submit those files for inclusion instead of making people guess what model.

Maybe if I knew to do that or how to do that, I would. I was just making a simple suggestion saying it didn't work. If there is a way of uploading, I would do so, but I am completely unfamiliar with GitHub as a whole.

Crsarmv7l, 2022-11-19:
> Maybe if I knew to do that or how to do that, I would. I was just making a simple suggestion saying it didn't work. If there is a way of uploading, I would do so, but I am completely unfamiliar with [...]

[image]

gsurkov, 2022-11-24:
@ALD53105 There already are instructions on adding other types of universal remotes [here](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/UniversalRemotes.md). It doesn't cover the TV part yet, as it requires a bit of refactoring, but it's pretty similar to audio players, so you might try to figure that out yourself (or wait until the proper instruction will be available, which is soon).

It won't cover making pull requests, however, but there is lots of info on that on the Internet.

closed 2022-11-24 by gsurkov