flipperdevices/flipperzero-firmware #2017: Build 0.71.1 failing

velezh, 2022-11-15:
### Describe the bug.

Build 0.71.1 failing on Mac
./fbt throws error (full listing in Logs section)

TypeError: Descriptors cannot not be created directly.

Using fbt (previous versions was built normally, for example 0.70.1)

```
Python 3.10.6

Аппаратное обеспечение:

  Название модели:	MacBook Air
  Идентификатор модели:	MacBookAir10,1
  Номер модели:	MGN63UA/A
  Чип:	Apple M1

Обзор системного ПО:

  Версия системы:	macOS 13.0.1 (22A400)
  Версия Kernel:	Darwin 22.1.0
```

### Reproduction

git pull
./fbt DEBUG=0

### Logs

```Text
vadimvelezh@MacBook-Air-vadim flipperzero-firmware % ./fbt DEBUG=0
Checking for tar..yes
Checking if downloaded toolchain tgz exists..no
Checking curl..yes
Downloading toolchain:
################################################################################################################# 100.0%
done
Removing old toolchain..done
Unpacking toolchain:
################################################################### 100.0%
done
Cleaning up..done
Submodule path 'assets/protobuf': checked out 'e5af96e08fea8351898f7b8c6d1e34ce5fd6cdef'
	ICONS	build/f7-firmware/assets/compiled/assets_icons.c
	PROTO	assets/protobuf/application.proto
	DOLPHIN	blocking
	DOLPHIN	internal
	APPS	build/f7-firmware/applications/applications.c
	CC	applications/main/archive/helpers/archive_apps.c
	CC	applications/main/archive/helpers/archive_browser.c
	CC	applications/main/archive/helpers/archive_favorites.c
2022-11-15 12:07:25,134 [INFO] Processing Dolphin sources
2022-11-15 12:07:25,134 [INFO] Loading data
2022-11-15 12:07:25,134 [INFO] Loading directory assets/dolphin/blocking
2022-11-15 12:07:25,135 [INFO] Loading meta from assets/dolphin/blocking/L0_NoDb_128x51/meta.txt
2022-11-15 12:07:25,135 [INFO] Loading meta from assets/dolphin/blocking/L0_SdBad_128x51/meta.txt
2022-11-15 12:07:25,135 [INFO] Loading meta from assets/dolphin/blocking/L0_SdOk_128x51/meta.txt
2022-11-15 12:07:25,136 [INFO] Loading meta from assets/dolphin/blocking/L0_Url_128x51/meta.txt
2022-11-15 12:07:25,136 [INFO] Loading meta from assets/dolphin/blocking/L0_NewMail_128x51/meta.txt
2022-11-15 12:07:25,137 [INFO] Packing
	CC	applications/main/archive/helpers/archive_files.c
2022-11-15 12:07:25,164 [INFO] Processing Dolphin sources
2022-11-15 12:07:25,165 [INFO] Loading data
2022-11-15 12:07:25,165 [INFO] Loading directory assets/dolphin/internal
2022-11-15 12:07:25,165 [INFO] Loading meta from assets/dolphin/internal/L1_Tv_128x47/meta.txt
2022-11-15 12:07:25,166 [INFO] Loading meta from assets/dolphin/internal/L1_BadBattery_128x47/meta.txt
2022-11-15 12:07:25,166 [INFO] Loading meta from assets/dolphin/internal/L1_NoSd_128x49/meta.txt
2022-11-15 12:07:25,166 [INFO] Packing
Traceback (most recent call last):
  File "/Users/vadimvelezh/Projects/flipperzero-firmware/./lib/nanopb/generator/nanopb_generator.py", line 51, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/vadimvelezh/Projects/flipperzero-firmware/./lib/nanopb/generator/nanopb_generator.py", line 72, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "/Users/vadimvelezh/Projects/flipperzero-firmware/lib/nanopb/generator/proto/__init__.py", line 23, in <module>
    from . import nanopb_pb2
  File "/Users/vadimvelezh/Projects/flipperzero-firmware/lib/nanopb/generator/proto/nanopb_pb2.py", line 35, in <module>
    _descriptor.EnumValueDescriptor(
  File "/Users/vadimvelezh/Projects/flipperzero-firmware/toolchain/x86_64-darwin/python/lib/python3.9/site-packages/google/protobuf/descriptor.py", line 755, in __new__
    _message.Message._CheckCalledFromGeneratedFile()
TypeError: Descriptors cannot not be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).

More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates
scons: *** [build/f7-firmware/assets/compiled/application.pb.c] Error 1
2022-11-15 12:07:30,052 [INFO] Complete
2022-11-15 12:07:30,052 [INFO] Complete

********** FBT ERRORS **********
build/f7-firmware/assets/compiled/application.pb.c: Error 1
```

hedger, 2022-11-15:
@drunkbatya have a look

bbock, 2022-11-15:
I had the same, updating the repo from a very old build. Solution: remove `lib/nanopb` to re-initialize submodule solved the problem for me.

velezh, 2022-11-15:
> I had the same, updating the repo from a very old build. Solution: remove `lib/nanopb` to re-initialize submodule solved the problem for me.

I tried and this works for me. Thanks a lot!

skotopes, 2022-12-01:
I'll close this issue then. We'll try to add additional check for this in future releases.

closed 2022-12-01 by skotopes