flipperdevices/flipperzero-firmware #2018: AZERTY not working as expected on badusb

baboufragile, 2022-11-15:
### Describe the bug.

When i use bad usb on a computer working with AZERTY keyboard it's not working, the flipper write like if it was in azerty, so would it be possible to check the keyboard language before writing something ? 

### Reproduction

change your language keyboard from QWERTY to AZERTY and launch a badusb program ( for me it's the demo for windows ),
it shouldn't work

### Target

badusb

skotopes, 2022-11-15:
This is known issue. Currently we don't support azerty and other keyboard layouts.

baboufragile, 2022-11-15:
okay, do you know if it's planned to add this support soon or not  ?

skotopes, 2022-11-15:
No plans on our side. There were some PRs but abandoned before I were able to merge them.

FelixLgr, 2022-11-19:
Someone told me to use `ALTSTRING` to solve this problem but he writes numbers instead.

skotopes, 2023-03-13:
Keyboard layout are now selectable in BadUSB config menu.

closed 2023-03-13 by skotopes