flipperdevices/flipperzero-firmware #2019: Weather station showing only one shot signal

michelinok, 2022-11-15:
### Describe the bug.

The weather station app receive/decode only one time the signal, to receive again you've to close the app and re-open it.

### Reproduction

Open weather station
Decode one sensor and wait

skotopes, 2022-11-15:
Can you provide sensor recording?

Skorpionm, 2022-11-15:
No, you are wrong, it receives and records 1 time if a NEW sensor is received. receiving REPEAT data from the same sensor UPDATE the data. Check, accept any new sensor, open it, and physically place the sensor in the refrigerator and see how the readings change

skotopes, 2022-12-01:
I'll close this issue. Please reopen and provide us RAW recording If you still experiencing this problem.

closed 2022-12-01 by skotopes